ALIGN-analoglayout/ALIGN-public #1079: Compiler To Generate Re-usable Constraints

soneryaldiz, 2022-04-21:
The auto-generated constraints are not always re-usable by the end user to copy and modify. User has to write constraints from scratch even for a simple OTA circuit. The compiler should generate constraints that are applicable on the original netlist and thereby re-usable by the end user.

soneryaldiz, 2022-04-21:
@kkunal1408 , can we scope the effort it takes and what the gaps are to achieve this?

kkunal1408, 2022-04-21:
After annotation, for the grouped blocks the individual block name information is not stored in the current database. 
e.g. two identified groups DP(MN7, MN8), DP(MN2, MN3) create only one subcircuit hierarchy DP(MN1, MN2) if the devices are equal.  Changing the database would take lots of effort. 
There are two tricks that I can think of now:
1. The name information is stored as a string x_MN7_MN8 which can be used to break it down. (It assumes names do not have a '_')
2.  The pre annotation database can be stored separately and names from this can be used for matching. 

We can do a brainstorming session on this to figure out the best way.

soneryaldiz, 2022-05-05:
@kkunal1408 , I think the value of user-consumable constraints is significantly higher than reducing redundancy in the subcircuit hierarchy. Furthermore, such a redundancy/duplication can be identified procedurally and be simplified before calling the primitive generator if needed.

soneryaldiz, 2022-05-05:
Per your example, constraints would look like:
```java
{"constraint": "GroupBlocks", "instances": ["mn7", "mn8"], "name": "<uuid1>", "generator":{<...>}},
{"constraint": "GroupBlocks", "instances": ["mn2", "mn3"], "name": "<uuid2>", "generator":{<...>}},
```

soneryaldiz, 2022-05-05:
Is there any other roadblock that you can think of?

kkunal1408, 2022-05-05:
No major roadblocks right now and the group block extension with generator constraint was done thinking this issue in mind. I will soon provide an update on this.
One minor roadblock is due to some name-based transformations during primitive generation (primitive/main.py) may fail in case the user changes the name. I will try to fix this with minimal disturbance but some PDK interfaces may break.

soneryaldiz, 2022-05-12:
@kkunal1408 , should we start a PR to do this?

kkunal1408, 2022-05-12:
I will do it by this weekend.

soneryaldiz, 2022-05-16:
Notes:
TODAY:
```
{
    "constraint": "GroupBlocks", 
    "instances": ["mn1", "mn2"], 
    "name": "<subcircuit name>",
}
```
After this PR:
```
{
    "constraint": "GroupBlocks", 
    "instances": ["mn1", "mn2"], 
    "name": "<instance name>",
    "template": "<subckt name>"   // Optional
}
```

closed 2022-05-26 by soneryaldiz