ALIGN-analoglayout/ALIGN-public #1093: Bulk testcases are failing

arvuce22, 2022-05-05:
I am unable to run any bulk test cases. Following is an error for "five_transistor_ota_Bulk" example. Any inputs to resolve this?
[image]

kkunal1408, 2022-05-05:
@Lastdayends based on the error message looks like the placer should be able to satisfy the constraints. Can you please verify once?

854768750, 2022-05-05:
okay

Lastdayends, 2022-05-05:
@kkunal1408 @arvuce22 @soneryaldiz 

It is a illegal placement from the view of the checker. May we relax the constraint first for Arvind to test the development? It might need some time to improve placer for a illegal solution.

soneryaldiz, 2022-05-05:
@arvuce22 , you can turn off auto-constraint generation to keep going. 
Both mn2/mn3 and mp4/mp5 have even number of legs. Therefore, the difference between the width of these two blocks should be zero or even multiple of poly pitch. I would revisit the primitive generator code. Our practice is always generate even width primitives so that they can be placed on a line of symmetry.

If you think that this is too limiting, we can soften the symmetry constraint checking.

arvuce22, 2022-05-05:
@Yaldiz, Soner ***@***.***> Thank you! I will try with these
modifications.

arvuce22, 2022-05-12:
Resolved

closed 2022-05-12 by arvuce22