ALIGN-analoglayout/ALIGN-public #1106: SameTemplate for Virtual Hierarchy

soneryaldiz, 2022-05-13:
Enhance compiler so that it is possible to specify SameTemplate for two separate GroupBlocks when they are equivalent. Compiler should throw assertion if the GroupBlocks do not match.

kkunal1408, 2022-05-14:
Right now since template names are unique, the compiler translates them to their instance names. What to do if constraints are defined based on these template names? Is it OK to modify the GroupBlocks constraint with two parameters one with template name and another instance name (concerned about larger impact)?

soneryaldiz, 2022-05-15:
@kkunal1408 , all constraints with few exceptions (AspectRatio, Boundary, ..) are on the instance names. What do you mean by constraints on template names? I am not quite following.

Let me describe a motivating example:
```javascript
[
  {"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "name": "grp1"},
  {"constraint": "GroupBlocks", "instances": ["mn3", "mn4"], "name": "grp2"},
  {"constraint": "SameTemplate", "instances": ["grp1", "grp2"]}
]
```
The constraints above could be auto-generated or user-provided. Any other placement constraint is still on the unique instance names `grp1` and `grp2`. Therefore, this shall imply no changes in the other constraints. The only changes should be in hierarchy naming/manipulation which is totally in compiler's control.

stevenmburns, 2022-05-16:
If we have a generator constraint we will probably only want to specify it for one of the `GroupBlocks` constraint.
We could do something like this to embrace the non-symmetry in the specification:
```
[
  {"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "name": "grp1", generator: { "name": "MOS", "parameters": {}}},
  {"constraint": "GroupBlocks", "instances": ["mn3", "mn4"], "name": "grp2", "same_template_as": "grp1"}
]
```
or, if specifying the template (abstract) name is important:
```
[
  {"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "name": "grp1", "template_name": "tn", generator: { "name": "MOS", "parameters": {}}},
  {"constraint": "GroupBlocks", "instances": ["mn3", "mn4"], "name": "grp2", "same_template_as": "grp1"}
]
```
or, if we could keep the specification symmetric (even though we will only write one generator statement), as:
```
[
  {"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "name": "grp1", generator: { "name": "MOS", "parameters": {}}},
  {"constraint": "GroupBlocks", "instances": ["mn3", "mn4"], "name": "grp2"},
  {"constraint": "SameTemplate", "instances": ["grp1", "grp2"], "template_name": "tb"}
]
```

soneryaldiz, 2022-05-16:
@kkunal1408 and I were just discussing this in #1109 .
A hacky way is to over-constrain that group (such as specifying a single aspect ratio) so that the placer can generate a single variant.
A better way is to introduce a new constraint that can limit the number of variant for a subcircuit. This will ensure all instances across hierarchy to have the same variant.