ALIGN-analoglayout/ALIGN-public #1109: Auto-generate SameTemplate

soneryaldiz, 2022-05-15:
```python
class ConfigureCompiler(SoftConstraint):
  same_template: bool = True
```
This flag auto-generates a `SameTemplate` constraint for every subcircuit in the netlist that has more than one instances.
During constraint translation to PnR, combine `SameTemplate` constraints with overlapping instances (union find).

soneryaldiz, 2022-05-15:
@kkunal1408 , do you want to do this?

kkunal1408, 2022-05-15:
What is all? Is it all matching subcircuits? Can you provide a test case for it?

soneryaldiz, 2022-05-15:
Here is an example. Given netlist:
```
.subckt poor_man_cascode d g s vcc vss
.ends
.subckt amplifier 
i1<0> d1 g1 s poor_man_cascode
i1<1> d1 g1 s poor_man_cascode
i1<2> d1 g1 s poor_man_cascode
.ends
```
Auto-generate:
```
{"constraint": "SameTemplate", "instances": ["i1<0>", "i1<1>", "i2<2>"]}
```

I will start a PR on Monday.

soneryaldiz, 2022-05-16:
Notes
- This feature is for physical subcircuits that are defined in the input netlist. This does not apply to virtual subcircuits that are defined via GroupBlocks.
- Would it be possible to use SameTemplate across hierarchy? It does not seme trivial to do in bottom up placement. It may require solving placement problem for each variant of the atomic subcircuit. One way could be to limit the number of variants for the critical subcircuit (through a new constraint that can limit how many variants a subcircuit can have).
```
.subckt poor_man_cascode
.ends
.subckt stage1
i0 ... poor_man_cascode
i1 ... poor_man_cascode
.ends
.subckt stage2
i0 ... poor_man_cascode
i1 ... poor_man_cascode
.ends
.subckt amplifier 
s1 ... stage1
s2 ... stage2
.ends
```
- When should this step be executed? After stack detection, after remove dummy hierarchy (subcircuit with a single instance), before group blocks.

closed 2022-06-29 by kkunal1408