ALIGN-analoglayout/ALIGN-public #1111: Several of the examples are not compiling

supertjhok, 2022-05-16:
Examples:

1) Five transistor OTA

schematic2layout.py ../examples/five_transistor_ota -p ../pdks/FinFET14nm_Mock_PDK/
align.main INFO : Reading netlist: /home/supertjhok/ALIGN-public/examples/five_transistor_ota/five_transistor_ota.sp subckt=FIVE_TRANSISTOR_OTA, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.user_const INFO : Reading constraint file: [PosixPath('/home/supertjhok/ALIGN-public/examples/five_transistor_ota/five_transistor_ota.const.json')]
align.compiler.compiler INFO : Completed topology identification.
align.pnr.main INFO : Running Place & Route for FIVE_TRANSISTOR_OTA
align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/cmdline.py", line 166, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/main.py", line 187, in schematic2layout
    variants = generate_pnr(topology_dir, primitive_dir, pdk_dir, pnr_dir, subckt, primitives=primitives, nvariants=nvariants, effort=effort,
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/main.py", line 291, in generate_pnr
    cap_map, cap_lef_s = cap_placer_driver(toplevel_args_d=toplevel_args_d, results_dir=None)
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/cap_placer.py", line 53, in cap_placer_driver
    assert required_caps == generated_caps, "Required capacitors {required_caps} different from generated_caps {generated_caps}."
AssertionError: Required capacitors {required_caps} different from generated_caps {generated_caps}.

2) Telescopic OTA

schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/
align.main INFO : Reading netlist: /home/supertjhok/ALIGN-public/examples/telescopic_ota/telescopic_ota.sp subckt=TELESCOPIC_OTA, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.user_const INFO : Reading constraint file: [PosixPath('/home/supertjhok/ALIGN-public/examples/telescopic_ota/telescopic_ota.const.json')]
align.compiler.compiler INFO : Completed topology identification.
constraint='generator' name='mos' parameters={'pattern': 'ncc', 'parallel_wires': {'DA': 1, 'DB': 1}}
align.pnr.main INFO : Running Place & Route for TELESCOPIC_OTA
align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/cmdline.py", line 166, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/main.py", line 187, in schematic2layout
    variants = generate_pnr(topology_dir, primitive_dir, pdk_dir, pnr_dir, subckt, primitives=primitives, nvariants=nvariants, effort=effort,
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/main.py", line 291, in generate_pnr
    cap_map, cap_lef_s = cap_placer_driver(toplevel_args_d=toplevel_args_d, results_dir=None)
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/cap_placer.py", line 53, in cap_placer_driver
    assert required_caps == generated_caps, "Required capacitors {required_caps} different from generated_caps {generated_caps}."
AssertionError: Required capacitors {required_caps} different from generated_caps {generated_caps}.

3) Telescoping OTA with guard ring

-> Same error

Latest pull from Github, Ubuntu 22.04, Python 3.10

kkunal1408, 2022-05-17:
Thanks for highligting. The supported Python version for ALIGN is python 3.8 and above. 

The error message is incomplete due to a typo in the line 53 "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/cap_placer.py".
Can you change that to 

```
    assert required_caps == generated_caps, f"Required capacitors {required_caps} different from generated_caps {generated_caps}."
```
and paste the assertion error message again?

supertjhok, 2022-05-17:
Thanks. Modified error message is as follows:

schematic2layout.py ../examples/telescopic_ota_guard_ring -p ../pdks/FinFET14nm_Mock_PDK/
align.main INFO : Reading netlist: /home/supertjhok/ALIGN-public/examples/telescopic_ota_guard_ring/telescopic_ota_guard_ring.sp subckt=TELESCOPIC_OTA_GUARD_RING, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.user_const INFO : Reading constraint file: [PosixPath('/home/supertjhok/ALIGN-public/examples/telescopic_ota_guard_ring/telescopic_ota_guard_ring.const.json')]
align.compiler.compiler INFO : Completed topology identification.
constraint='generator' name='mos' parameters={'pattern': 'ncc', 'parallel_wires': {'DA': 1, 'DB': 1}}
align.pnr.main INFO : Running Place & Route for TELESCOPIC_OTA_GUARD_RING
align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/cmdline.py", line 166, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/main.py", line 187, in schematic2layout
    variants = generate_pnr(topology_dir, primitive_dir, pdk_dir, pnr_dir, subckt, primitives=primitives, nvariants=nvariants, effort=effort,
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/main.py", line 291, in generate_pnr
    cap_map, cap_lef_s = cap_placer_driver(toplevel_args_d=toplevel_args_d, results_dir=None)
  File "/home/supertjhok/ALIGN-public/work/general/lib/python3.10/site-packages/align/pnr/cap_placer.py", line 53, in cap_placer_driver
    assert required_caps == generated_caps, f"Required capacitors {required_caps} different from generated_caps {generated_caps}."
AssertionError: Required capacitors set() different from generated_caps {'CAP_CC_3_5'}.

supertjhok, 2022-05-17:
Thanks. The problem turned out to be point #2. Cleaning out the work area solves the problem.

closed 2022-05-18 by kkunal1408