ALIGN-analoglayout/ALIGN-public #1113: About training database of ML

Arvin1000, 2022-05-17:
Hi, I read the paper of ALIGN, knowing that it uses machine learning to train some circuits or primitives.

But I didn't find where is the training database and how you train in your code.

Could you please tell where to find them, thanks very much!

kkunal1408, 2022-05-18:
Hello @Arvin1000 These are the OTA topologies used for training https://github.com/ALIGN-analoglayout/ALIGN-public/tree/master/CircuitsDatabase/Unsized_Netlists/Low%20frequency%20analog/OTA_topologies

Arvin1000, 2022-05-26:
a...then how you train with ML?
I researched your paper GANA,but didn't find codes either in the paper or here.
so....whether or not you open your code of training?How can we find them?
Thanks very much!

kkunal1408, 2022-11-03:
Will provide it later.

closed 2022-11-03 by kkunal1408