ALIGN-analoglayout/ALIGN-public #1152: Error in run  adder

AloysDu, 2022-07-10:
When I run the align to run adder in Example, I get an error, but the layout is generated. Can these errors and warnings be ignored ?
The command and error information are as follows.

schematic2layout.py ../examples/adder -p ../pdks/FinFET14nm_Mock_PDK/

 I get this log:

align.pnr.router INFO : Starting top_down routing on ADDER 0 restricted to None
PnR.router.Router.RouteWork INFO : GcellGlobalRouter: ADDER
PnR.router.Router.RouteWork INFO : GcellDetailRouter: ADDER
PnR.router.Router.RouteWork INFO : Create power grid: ADDER
PnR.router.Router.RouteWork INFO : Power routing ADDER
align.cell_fabric.remove_duplicates WARNING : Layer Cboundary not in OrderedDict([('M1', 'v'), ('M2', 'h'), ('V1', '*'), ('M3', 'v'), ('V2', '*'), ('M4', 'h'), ('V3', '*'), ('M5', 'v'), ('V4', '*'), ('M6', 'h'), ('V5', '*'), ('M7', 'v'), ('V6', '*'), ('M8', 'h'), ('V7', '*'), ('M9', 'v'), ('V8', '*'), ('M10', 'h'), ('V9', '*'), ('M11', 'v'), ('V10', '*'), ('M12', 'h'), ('V11', '*'), ('M13', 'v'), ('V12', '*'), ('M14', 'h'), ('V13', '*'), ('M15', 'v'), ('V14', '*'), ('Poly', 'v'), ('Fin', 'h'), ('Active', 'h'), ('Rvt', 'h'), ('Lvt', 'h'), ('Hvt', 'h'), ('Slvt', 'h'), ('Lisd', 'v'), ('Pc', 'h'), ('Pb', 'h'), ('V0', '*')])
align.cell_fabric.remove_duplicates WARNING : Layer Cboundary not in OrderedDict([('M1', 'v'), ('M2', 'h'), ('V1', '*'), ('M3', 'v'), ('V2', '*'), ('M4', 'h'), ('V3', '*'), ('M5', 'v'), ('V4', '*'), ('M6', 'h'), ('V5', '*'), ('M7', 'v'), ('V6', '*'), ('M8', 'h'), ('V7', '*'), ('M9', 'v'), ('V8', '*'), ('M10', 'h'), ('V9', '*'), ('M11', 'v'), ('V10', '*'), ('M12', 'h'), ('V11', '*'), ('M13', 'v'), ('V12', '*'), ('M14', 'h'), ('V13', '*'), ('M15', 'v'), ('V14', '*'), ('Poly', 'v'), ('Fin', 'h'), ('Active', 'h'), ('Rvt', 'h'), ('Lvt', 'h'), ('Hvt', 'h'), ('Slvt', 'h'), ('Lisd', 'v'), ('Pc', 'h'), ('Pb', 'h'), ('V0', '*')])
align.cell_fabric.remove_duplicates ERROR : Found errors: SHORT: 0 OPEN: 0 DIFFERENT WIDTH: 3
align.cell_fabric.postprocess ERROR : Found errors: POSTPROCESSING: 501
align.pnr.main INFO : OUTPUT json at /home/aloysdu/ALIGN-public/workspace/adder/3_pnr/ADDER_0.json
align.pnr.main INFO : OUTPUT gds.json /home/aloysdu/ALIGN-public/workspace/adder/3_pnr/ADDER_0.python.gds.json
Use KLayout to visualize the generated GDS: /home/aloysdu/ALIGN-public/workspace/adder/ADDER_0.gds
Use KLayout to visualize the python generated GDS: /home/aloysdu/ALIGN-public/workspace/adder/ADDER_0.python.gds

kkunal1408, 2022-07-11:
Yes, these errors are fine. These are due to different width nets used for capacitors. Our inbuilt DRC checker assumes uniform width and thus shows a DRC but these are not present when we use a real Calibre DRC checker.

AloysDu, 2022-07-11:
> Yes, these errors are fine. These are due to different width nets used for capacitors. Our inbuilt DRC checker assumes uniform width and thus shows a DRC but these are not present when we use a real Calibre DRC checker.

get it. thanks

closed 2022-08-04 by kkunal1408