ALIGN-analoglayout/ALIGN-public #1153: Assert False, Unmatched generator for this instance

AloysDu, 2022-07-11:
Hello,
I converted some simple SPICE netlists according to the format in Align example, and then ran the layout and wiring to test the efficiency of Align, and most of them were successful. However, I noticed that there was error in a small part of them. I looked at the SP file carefully and compared it with other successful cases,It told me to fix netlist But I can't find the reason why the m11 statement is wrong.It's so confusing to me. 

### the sp file as follow:

```
.subckt 5_16 GNDA VDDA Vbias1 Vbias2 vin vip vout
m13 net27 net18 VDDA VDDA pmos_rvt W=600n L=800n M=1 nfin=2 nf=2
m0 net18 net14 VDDA VDDA pmos_rvt W=600n L=800n M=1 nfin=2 nf=2
m1 net14 net14 VDDA VDDA pmos_rvt W=600n L=800n M=1 nfin=2 nf=2
m11 VDDA net27 vout VDDA nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m9 net26 Vbias2 GNDA GNDA nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m10 net27 Vbias1 net26 net26 nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m2 net14 vip net13 net13 nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m3 net18 vin net13 net13 nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m4 net13 Vbias1 net34 GNDA nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m5 net34 Vbias2 GNDA GNDA nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m6 vout Vbias1 net22 net22 nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
m7 net22 Vbias2 GNDA GNDA nmos_rvt W=600n L=600n M=1 nfin=2 nf=2
C0 net18 vout 1p
.ENDS
```

### I get this log:

```
align.main INFO : Reading netlist: /home/aloysdu/ALIGN-public/examples/netlist_pending/5_16/5_16.sp subckt=5_16, flat=0
 align.compiler.compiler INFO : Starting topology identification...
 align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/aloysdu/.local/lib/python3.8/site-packages/align/cmdline.py", line 166, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/aloysdu/.local/lib/python3.8/site-packages/align/main.py", line 163, in schematic2layout
    primitive_lib = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir)
  File "/home/aloysdu/.local/lib/python3.8/site-packages/align/compiler/compiler.py", line 35, in generate_hierarchy
    primitives = PrimitiveLibrary(ckt_data, pdk_dir).gen_primitive_collateral()
  File "/home/aloysdu/.local/lib/python3.8/site-packages/align/compiler/gen_abstract_name.py", line 53, in gen_primitive_collateral
    self.gen_primitive_def(ele)
  File "/home/aloysdu/.local/lib/python3.8/site-packages/align/compiler/gen_abstract_name.py", line 122, in gen_primitive_def
    assert False, f"Unmatched generator for this instance {element}, please fix netlist "
AssertionError: Unmatched generator for this instance model='NMOS_RVT' name='M11' pins={'D': 'VDDA', 'G': 'NET27', 'S': 'VOUT', 'B': 'VDDA'} parameters={'W': '6.000000000000001E-07', 'L': '6.000000000000001E-07', 'NFIN': '1', 'NF': '2', 'M': '1', 'PARALLEL': '1', 'STACK': '1'} abstract_name=None, please fix netlist 
```

Can anyone give me some tips?
Thank you in advance.

kkunal1408, 2022-08-04:
Which PDK are you using? It seems like the nmos_rvt device name is not defined for the specific pdk.
It needs to be defined in pdks/<pdk_name>models.sp e..g "pdks/FinFET14nm_Mock_PDK/models.sp"

kkunal1408, 2022-08-11:
@AloysDu let me know if this worked for you.

kkunal1408, 2022-11-03:
Closing as no response.

closed 2022-11-03 by kkunal1408