ALIGN-analoglayout/ALIGN-public #1161: AssertionError: incorrect symmetry branch

parkerluxu, 2022-09-13:
Hello, I got this error when I want to generate my layout from the schematic.
 my spice file is like the following:

```
.subckt INVD2BWP35P140 i zn vdd vss
xmm1 zn i vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm1_2 zn i vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm0 zn i vdd vdd pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
xmm0_2 zn i vdd vdd pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
.ends INVD2BWP35P140

.subckt ISOSRHID2BWP35P140 iso i z vdd vdds vss
xmm8 net50 iso vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm9 net50 i vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm0 z net50 vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm0_2 z net50 vss vss nmos_rvt w=280e-9 l=35e-9 nfin=4 nf=2
xmm7 net36 iso vdds vdds pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
xmm3 net50 i net36 vdds pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
xmm1 z net50 vdds vdds pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
xmm1_2 z net50 vdds vdds pmos_rvt w=340e-9 l=35e-9 nfin=4 nf=2
.ends ISOSRHID2BWP35P140

.subckt rxanlg_ls_sum_data data_en data_en_ana dvdd ibias_con<3> ibias_con<2> ibias_con<1> ibias_con<0> ibias_con_ana<3> ibias_con_ana<2> ibias_con_ana<1> ibias_con_ana<0> por por_ana vddio vref01_cal<6> vref01_cal<5> vref01_cal<4> vref01_cal<3> vref01_cal<2> vref01_cal<1> vref01_cal<0> vref01_cal_ana<6> vref01_cal_ana<5> vref01_cal_ana<4> vref01_cal_ana<3> vref01_cal_ana<2> vref01_cal_ana<1> vref01_cal_ana<0> vss vssio
xi29 por net07 dvdd vss INVD2BWP35P140
xi28 por net015 dvdd vss INVD2BWP35P140
xi27 por net014 dvdd vss INVD2BWP35P140
xi26<6> net032<0> vref01_cal_ana<6> vddio vssio INVD2BWP35P140
xi26<5> net032<1> vref01_cal_ana<5> vddio vssio INVD2BWP35P140
xi26<4> net032<2> vref01_cal_ana<4> vddio vssio INVD2BWP35P140
xi26<3> net032<3> vref01_cal_ana<3> vddio vssio INVD2BWP35P140
xi26<2> net032<4> vref01_cal_ana<2> vddio vssio INVD2BWP35P140
xi26<1> net032<5> vref01_cal_ana<1> vddio vssio INVD2BWP35P140
xi26<0> net032<6> vref01_cal_ana<0> vddio vssio INVD2BWP35P140
xi23<6> vref01_cal<6> net031<0> dvdd vss INVD2BWP35P140
xi23<5> vref01_cal<5> net031<1> dvdd vss INVD2BWP35P140
xi23<4> vref01_cal<4> net031<2> dvdd vss INVD2BWP35P140
xi23<3> vref01_cal<3> net031<3> dvdd vss INVD2BWP35P140
xi23<2> vref01_cal<2> net031<4> dvdd vss INVD2BWP35P140
xi23<1> vref01_cal<1> net031<5> dvdd vss INVD2BWP35P140
xi23<0> vref01_cal<0> net031<6> dvdd vss INVD2BWP35P140
xi22<3> net030<0> ibias_con_ana<3> vddio vssio INVD2BWP35P140
xi22<2> net030<1> ibias_con_ana<2> vddio vssio INVD2BWP35P140
xi22<1> net030<2> ibias_con_ana<1> vddio vssio INVD2BWP35P140
xi22<0> net030<3> ibias_con_ana<0> vddio vssio INVD2BWP35P140
xi20<3> ibias_con<3> net029<0> dvdd vss INVD2BWP35P140
xi20<2> ibias_con<2> net029<1> dvdd vss INVD2BWP35P140
xi20<1> ibias_con<1> net029<2> dvdd vss INVD2BWP35P140
xi20<0> ibias_con<0> net029<3> dvdd vss INVD2BWP35P140
xi19 data_en net028 dvdd vss INVD2BWP35P140
xi16 net011 data_en_ana vddio vssio INVD2BWP35P140
xi24<6> net031<0> net015 net032<0> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<5> net031<1> net015 net032<1> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<4> net031<2> net015 net032<2> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<3> net031<3> net015 net032<3> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<2> net031<4> net015 net032<4> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<1> net031<5> net015 net032<5> vddio dvdd vssio ISOSRHID2BWP35P140
xi24<0> net031<6> net015 net032<6> vddio dvdd vssio ISOSRHID2BWP35P140
xi21<3> net029<0> net07 net030<0> vddio dvdd vssio ISOSRHID2BWP35P140
xi21<2> net029<1> net07 net030<1> vddio dvdd vssio ISOSRHID2BWP35P140
xi21<1> net029<2> net07 net030<2> vddio dvdd vssio ISOSRHID2BWP35P140
xi21<0> net029<3> net07 net030<3> vddio dvdd vssio ISOSRHID2BWP35P140
xi18 net028 net014 net011 vddio dvdd vssio ISOSRHID2BWP35P140
xi14<6> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<5> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<4> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<3> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<2> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<1> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
xi14<0> por por por_ana vddio dvdd vss ISOSRHID2BWP35P140
.ends rxanlg_ls_sum_data
```

I can get the layout with only INVD2BWP35P140 or ISOSRHID2BWP35P140. Are some issues in my design schematic?
the error message is the following:
```
raceback (most recent call last):
  File "/home/luxu/ALIGN-public/align/cmdline.py", line 166, in parse_args
    return schematic2layout(**vars(arguments))
  File "/home/luxu/ALIGN-public/align/main.py", line 163, in schematic2layout
    primitive_lib = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir)
  File "/home/luxu/ALIGN-public/align/compiler/compiler.py", line 36, in generate_hierarchy
    constraint_generator(ckt_data)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 244, in constraint_generator
    FindConst(subckt)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 268, in FindConst
    match_pairs = FindSymmetry(subckt, stop_points)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 227, in FindSymmetry
    recursive_start_points(graph, match_pairs, traversed, port1, port2, ports_weight)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 193, in recursive_start_points
    compare_nodes(G, match_pairs, pair, traversed, node1, node2, ports_weight)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 171, in compare_nodes
    compare_nodes(G, match_pairs, match_pair, traversed.copy(), nbr1, nbr2, ports_weight)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 171, in compare_nodes
    compare_nodes(G, match_pairs, match_pair, traversed.copy(), nbr1, nbr2, ports_weight)
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 171, in compare_nodes
    compare_nodes(G, match_pairs, match_pair, traversed.copy(), nbr1, nbr2, ports_weight)
  [Previous line repeated 1 more time]
  File "/home/luxu/ALIGN-public/align/compiler/find_constraint.py", line 77, in compare_nodes
    assert not match_pair.get("array_start_point", False), f"incorrect symmetry branch"
AssertionError: incorrect symmetry branch
```

soneryaldiz, 2022-09-16:
@parkerluxu auto-constraint generation is failing. Please try turning it off by adding a constraint:
```javascript
{"constraint": "ConfigureCompiler", "auto_constraint": false, "propagate": true, "merge_parallel_devices": false},
```
cc'ing @kkunal1408 .

parkerluxu, 2022-09-18:
Thanks, the problem has been solved by adding this constraint.

closed 2022-11-03 by kkunal1408