ALIGN-analoglayout/ALIGN-public #1162: Check that layers have unique GdsLayerNo in layers.json

soneryaldiz, 2022-09-15:
> > > For `test_ru_m1vt_h`, it is because "VT" is not recognized. So it is considered as "V1". Let me add VT definition in PnR
> > 
> > 
> > @854768750 , it is not a good idea to hard-code VT as a via name. All vias are already defined in layers.json. How about finding all vias from layers.json?
> 
> via name is not hard coded. Can we have different GdsLayerNo for VT and V1? Not they all have 201

Yes, we can and should have different GdsLayerNo. I will start an issue to address this later. I will fix the layer number right away.

_Originally posted by @soneryaldiz in https://github.com/ALIGN-analoglayout/ALIGN-public/issues/1155#issuecomment-1248720509_

srini229, 2023-10-02:
@soneryaldiz is this still required?

soneryaldiz, 2023-10-02:
No required.

closed 2023-10-02 by kkunal1408