ALIGN-analoglayout/ALIGN-public #1181: Template names are appended with a hash

kkunal1408, 2022-09-29:
If user is specifying a template (i.e., subcircuit) name in a group block, why append a hash to it?
1. hash is an indirect way of storing the subcircuit information (sizing + constraints). This helps in comparing two blocks for symmetry.
2. This helps in the reduction of storing duplicate hierarchies.

This PR will explore the impact of removing the hash.

kkunal1408, 2022-09-29:
@soneryaldiz hash removal will be targeted here.

kkunal1408, 2022-11-03:
Hashes are important to identify identical subcircuits. Name-based matches may not work for virtual hierarchies in case the constraints are different for two group blocks.

soneryaldiz, 2022-11-10:
Closing as won't do as user can directly specify constraints for a virtual hierarchy:
```
{ 
	"constraint": "GroupBlocks",
	"instance_name": str,
	"instances": [],
	"template_name": diff_pair,
	"constraints": [
	
	]
}
```

closed 2022-11-10 by soneryaldiz