ALIGN-analoglayout/ALIGN-public #1188: What is the difference between HPWL_extend vs HPWL_ILP?

soneryaldiz, 2022-10-10:
What could cause HPWL_extend to be zero or negative?

soneryaldiz, 2022-10-10:
Examples:
```
2022-10-10 11:47:38 PnR.placer.ILP_solver.GenerateValidSolution DEBUG : ILP__HPWL_compare : HPWL_extend=24532 HPWL_ILP=34285
...
2022-10-10 11:47:39 PnR.placer.ILP_solver.GenerateValidSolution DEBUG : ILP__HPWL_compare : HPWL_extend=0 HPWL_ILP=3971
```

srini229, 2022-10-10:
HPWL_ILP is the HPWL evaluated using the terms used in the cost function of the ILP. It uses the bounding box of the pins.

From the code, HPWL_extend appears to be iterating over all the shapes of a pin to find the HPWL. HPWL_extend is set to 0 for a floating pin(a pin at a hierarchy with no connection).

854768750, 2022-10-10:
HPWL_extend considers the pin shape 
HPWL_ILP only takes center point of pin

soneryaldiz, 2022-10-10:
> HPWL_extend considers the pin shape HPWL_ILP only takes center point of pin

Are you sure? The code indicates the opposite.

854768750, 2022-10-10:
HPWL_ILP also considers pin shape, using variable values from ILP solver
HPWL is center point only

soneryaldiz, 2022-10-13:
Some more examples:
```
2022-10-12 15:07:10 align.pnr.placer WARNING : hpwl: locally computed from netlist 32284, placer computed 25484 differ for DIFFAMP_FC_N_PG0_0!
2022-10-12 15:07:10 align.pnr.placer WARNING : hpwl: locally computed from netlist 34444, placer computed 26924 differ for DIFFAMP_FC_P_PG0_0!
2022-10-12 15:07:11 align.pnr.placer WARNING : hpwl: locally computed from netlist 4353, placer computed -1487 differ for OUTP_59570252_PG0_0!
2022-10-12 15:07:11 align.pnr.placer WARNING : hpwl: locally computed from netlist 12182, placer computed 4662 differ for DIFFAMP_NBIAS_LVT_PG0_0!
2022-10-12 15:07:11 align.pnr.placer WARNING : hpwl: locally computed from netlist 28232, placer computed 18192 differ for PRIMITIVE_7025008_PG0_0!
```

soneryaldiz, 2022-10-13:
Discussed in the meeting. Will start a PR to find a reproducing case. May need to build in debug mode to catch assertions or throw.

srini229, 2022-10-13:
I added an `exit()` locally to check for scenarios where HPWL_ILP and HPWL_extend do not match.
These three tests failed:
```
tests/pdk/finfet_pdk/test_annotation.py::test_dcl
tests/pdk/finfet_pdk/test_miscellaneous.py::test_floating_power_pin
tests/pdk/finfet_pdk/test_placement_grid.py::test_bias_generator
```

soneryaldiz, 2022-10-13:
> I added an `exit()` locally to check for scenarios where HPWL_ILP and HPWL_extend do not match. These three tests failed:
>
> ```
> tests/pdk/finfet_pdk/test_annotation.py::test_dcl [...]

Could you please start a PR?

soneryaldiz, 2022-10-13:
Is it possible to use HPWL_ILP and drop HWPL_extend? 
Is there a reason to calculate HPWL_extend again after an ILP solution?

srini229, 2022-10-13:
> Is it possible to use HPWL_ILP and drop HWPL_extend? 
>Is there a reason to calculate HPWL_extend again after an ILP solution?

If we ensure that we are doing the right thing in ILP, we can drop HPWL_extend.
Since we got rid of the compaction routine, there is no change to placement after ILP.
This means HPWL_ILP has to match the HPWL_extend excepting the scenario when there is no ILP called.
Which is for the trivial case of a single instance hierarchy with no PlaceOnGrid constraints.

854768750, 2022-10-13:
For this test case `test_dcl`, the net `VBIAS1` is only connected to one pin `X_MP33:S`. 
As there is only one pin in this net, HPWL_ILP skips this net, but HPWL_extend keeps this net.
Similar for net `V2` and net `V4`.
@soneryaldiz Do you want to count HPWL for these one-pin nets?

854768750, 2022-10-13:
This PR https://github.com/ALIGN-analoglayout/ALIGN-public/pull/733 tells why we added so many versions of HPWL.
HPWL: center points only, of all nets (regardless of one-pin net or n-pin net)
HPWL_extend: consider bounding box of pins, of all nets (regardless of one-pin net or n-pin net)
HPWL_extend_terminal: consider bounding box of pins, of terminal nets (at least one terminal in the net). exclude internal nets without terminal connection
HPWL_extend_wo_terminal: consider bounding box of pins, of internal nets (HPWL_extend - HPWL_extend_terminal)

854768750, 2022-10-13:
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/f742c81b32ed4782b2d2651ceb8a0012ec133e52/align/pnr/hpwl.py#L120 
The checker seems to have different net selection criteria.

soneryaldiz, 2022-10-13:
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/f742c81b32ed4782b2d2651ceb8a0012ec133e52/align/pnr/placer.py#L126

@854768750 , `skip_globals` is set to `True` which implies power and ground rails are excluded from the calculation.

soneryaldiz, 2022-10-13:
Reading #733 , `HPWL_extend` is added per @stevenmburns 's request. I suggest to keep only `HPWL_ILP` which by definition is supposed to match `HPWL_extend` and use it as the HPWL. Is there a reason to keep any other HPWL definition around? 

One quick experiment is to expose `HWPL_ILP` to `placer.py` and check if they ever mismatch.

854768750, 2022-10-13:
HPWL_ILP and HPWL_extend are different, when one-pin net exists.
HPWL_ILP does not consider one-pin net. HPWL_extend does.

soneryaldiz, 2022-10-14:
https://github.com/ALIGN-analoglayout/ALIGN-public/blob/f742c81b32ed4782b2d2651ceb8a0012ec133e52/align/pnr/hpwl.py#L157

The checker above also includes single-pin nets. @srini229 , is it straightforward to include single-terminal nets in HPWL_ILP?

854768750, 2022-10-14:
> https://github.com/ALIGN-analoglayout/ALIGN-public/blob/f742c81b32ed4782b2d2651ceb8a0012ec133e52/align/pnr/hpwl.py#L157
> 
> The checker above also includes single-pin nets. @srini229 , is it straightforward to include single-terminal nets in HPWL_ILP?

single-pin net is included in https://github.com/ALIGN-analoglayout/ALIGN-public/pull/1191/commits/62a20c6872379e082c2e5cc39abfad244366c601

closed 2022-10-22 by srini229