ALIGN-analoglayout/ALIGN-public #1194: Why is compiler generating a single instance hierarchy transistors that are identified?

soneryaldiz, 2022-10-21:
Consider two scenarios:
```python
# case 1
{"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "instance_name": "xtail"}
# case 2
{"constraint": "GroupBlocks", "instances": ["mn1", "mn2"], "instance_name": "xtail", "generator": { "name": "MOS"}}
```

Case 1 results in a single instance virtual hierarchy while case 2 does not and treated as a simple primitive. What is the reason for creating this redundant, single instance hierarchy?

kkunal1408, 2022-11-03:
Both cases have two instances MN1 and MN2 ? The difference between these is case one will be placed and routed by PnR e.g. an inverter and the second case will be generated by the primitive generator e.g., a differential pair.

soneryaldiz, 2022-11-10:
After subcircuit identification, case 1 is reduced to a single instance. Is it possible to detect this and eliminate this redundant hierarchy?

soneryaldiz, 2022-11-10:
PnR only stores a single solution per cost. Therefore, mirrored version of a block will not be returned as a unique variant.