ALIGN-analoglayout/ALIGN-public #1200: ImportError: cannot import name 'PnR'

uduse, 2022-10-28:
I am getting this error after following the installation guide and run `pytest -vv`. I tried both Linux and MacOS and this exact problem happens in both of them.

```
(general) vscode ➜ /workspaces/align_sandbox/ALIGN-public (master) $ pytest -vv
ImportError while loading conftest '/workspaces/align_sandbox/ALIGN-public/conftest.py'.
/Users/Uduse/GoogleDrive/Coding/align_sandbox/ALIGN-public/conftest.py:2: in <module>
    ???
align/__init__.py:6: in <module>
    from .main import schematic2layout
align/main.py:13: in <module>
    from .pnr import generate_pnr
align/pnr/__init__.py:1: in <module>
    from .checkers import *
align/pnr/checkers.py:7: in <module>
    from .router import NType
align/pnr/router.py:8: in <module>
    from .. import PnR
E   ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import) (/workspaces/align_sandbox/ALIGN-public/align/__init__.py)
```

stevenmburns, 2022-10-28:
Can you give me more details of what you did to install the tool and paste in the log file?
Can you follow the instructions in README.md (Step 0 through Step 4) then let us know where it first breaks?
Does Step 4 work for you, but not `pytest -vv`.

uduse, 2022-10-28:
@stevenmburns Thank you for your support 😃  I followed the exact steps 0, 1, 2, 3a. Step 4 works, I am able to generate layout, but the test never passes. The test is the only case where it breaks.

stevenmburns, 2022-10-28:
What version of Python are you using?
You seem to be in the "general" virtualenv.
Can you start up python in the virtualenv and run these three commands (the first three in `conftest.py` and tell me what you get:
```
import pytest
from align.utils import logmanager
logmanager.reconfigure_loglevels(console_level='WARNING')
```

uduse, 2022-10-30:
I'm using Python 3.8.15. The `general` virtualenv is suggested by the installation guide so I followed it verbatim.
```
(general) vscode ➜ /workspaces/align_sandbox $ python
Python 3.8.15 (default, Oct 26 2022, 04:07:55) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pytest
>>> from align.utils import logmanager
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/workspaces/align_sandbox/ALIGN-public/align/__init__.py", line 6, in <module>
    from .main import schematic2layout
  File "/workspaces/align_sandbox/ALIGN-public/align/main.py", line 13, in <module>
    from .pnr import generate_pnr
  File "/workspaces/align_sandbox/ALIGN-public/align/pnr/__init__.py", line 1, in <module>
    from .checkers import *
  File "/workspaces/align_sandbox/ALIGN-public/align/pnr/checkers.py", line 7, in <module>
    from .router import NType
  File "/workspaces/align_sandbox/ALIGN-public/align/pnr/router.py", line 8, in <module>
    from .. import PnR
  File "/workspaces/align_sandbox/ALIGN-public/align/PnR/__init__.py", line 1, in <module>
    from .checkers import *
  File "/workspaces/align_sandbox/ALIGN-public/align/PnR/checkers.py", line 7, in <module>
    from .router import NType
  File "/workspaces/align_sandbox/ALIGN-public/align/PnR/router.py", line 11, in <module>
    from .build_pnr_model import gen_DB_verilog_d
  File "/workspaces/align_sandbox/ALIGN-public/align/PnR/build_pnr_model.py", line 14, in <module>
    NType = PnR.NType
AttributeError: partially initialized module 'align.PnR' has no attribute 'NType' (most likely due to a circular import)
>>> logmanager.reconfigure_loglevels(console_level='WARNING')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'logmanager' is not defined
```

stevenmburns, 2022-10-31:
It seems that maybe PNR was not built correctly (a C++ extension) but I haven't seen this error message before.
You could try removing the _skbuild directory and trying all the steps over again.

I tried a reinstall from scratch on Ubuntu 18.04 and am not able to reproduce this issue.
If you let me know what OS you are using, I can make a Dockerfile that will document all the steps needed to install ALIGN on that OS. Then we have an exact flow we look at and see what problems might be occurring.

uduse, 2022-10-31:
@stevenmburns Thank you for your help, I really appreciate it 🙇‍♂️ I tried it both natively on my MacOs and inside of a Linux docker. This exact same problem happen in both of them.

stevenmburns, 2022-10-31:
Which linux do you want me to build a Dockerfile for? Ubuntu or Centos and which version.

uduse, 2022-10-31:
@stevenmburns Ubuntu will be great! Thanks a lot!

stevenmburns, 2022-10-31:
Ok, here is a Dockerfile that correctly builds align on Ubuntu 20.04
```
FROM ubuntu:20.04

RUN apt update \
    && DEBIAN_FRONTEND=noninteractive TZ=US/Pacific apt-get install -y tzdata \
    && apt install -yq build-essential git cmake \
    && apt install -yq python3 python3-venv python3-pip

RUN git clone --branch=master --depth=1 --single-branch https://github.com/ALIGN-analoglayout/ALIGN-public.git

WORKDIR /ALIGN-public

RUN python3 -m venv .venv

RUN /bin/bash -c "source .venv/bin/activate \
    && pip install --upgrade pip wheel \
    && pip install setuptools wheel pybind11 numpy scikit-build cmake ninja \
    && pip install -v -e .[test] --no-build-isolation \
    && pip install -v --no-build-isolation -e . --no-deps --install-option='--build-type=RelWithDebInfo' --install-option='-DBUILD_TESTING=ON'"
```
To build the container, in an empty directory try:
```
docker build -f <dockerfile name> .
```
You can then run the tests using:
```
docker run -it <image id>
```
and then in the container:
```
source .venv/bin/activate
export ALIGN_HOME=`pwd`
mkdir work
export ALIGN_WORK_DIR=$ALIGN_HOME/work
pytest -vv -n 8
```

stevenmburns, 2022-11-01:
I've upload an image to docker hub, so you can use that directly without building it yourself:
```
docker run -it stevenmburns/align
```

uduse, 2022-11-01:
@stevenmburns I have tried the image on the docker hub and I made through the tests without any problem. Thanks a lot! 🙇

closed 2022-11-01 by uduse