ALIGN-analoglayout/ALIGN-public #1204: ALIGN to work with open pdk

ChrisZonghaoLi, 2022-11-05:
Greetings,

So this is rather a very general question. Does anyone here ever try to use ALIGN to work with some open pdks such as skywater130? The reason I am asking is I intend to actually do so and I assume the biggest hump here is to (probably very labor intensive but should be straightforward) have all skywater130 design rules written in the JASON file (similar to that mock 14nm FInFET). Is this a good start? Or any advice here? Or anyone here is already doing so and probably we can co-operate (just do not want to do something and then it turns our somebody already done it or finish most of it)? Thanks.

arvuce22, 2022-11-10:
Its already done: You can follow the following repo: https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130

closed 2022-11-10 by arvuce22

ChrisZonghaoLi, 2022-11-10:
Affirmative. I will follow that repo and see how it goes. Thanks.