ALIGN-analoglayout/ALIGN-public #1207: Confusion on Inductors with FinFet PDK

ryancor, 2022-11-09:
We are allegedly allowed to use inductors in our spice files because I can see it in the models.sp file in that directory so I added it in like:

“L1 vps gnd inductor ind=1 parallel=1 value=1 stack=1” 

and it just keeps erroring out with “unmatched generator for this instance model=“INDUCTOR” name=L1 pins=etc etc… abstract_name=None, please fix netlist”

arvuce22, 2022-11-10:
We do not have a cell generator which can generate inductor layouts. Therefore, it is giving you an error. You can remove inductor instances from your netlist and try.

closed 2022-11-10 by arvuce22

ryancor, 2022-11-10:
gotcha okay, yea it worked before i added it just curious if there wasn't a generator for it yet. Any plans for it in the future?

ryancor, 2022-11-12:
@arvuce22 also any plans to add a cell generator for diodes?