ALIGN-analoglayout/ALIGN-public #1220: Multi-cut via not enclosed properly

srini229, 2022-11-16:
In the layout of an inverter generated with a modified layers.json, a few cuts of the V4 generated by the router are not properly enclosed.
Below is the layout of the inverter
[image]

Zoomed in version of the via:
[image]

To reproduce, use the branch `fix/multi_cut_via` and run:
`schematic2layout.py -p ${ALIGN_HOME}/pdks/FinFET14nm_Mock_PDK ${ALIGN_HOME}/examples/inverter`

854768750, 2022-11-16:
Thanks for letting me know these. I will take look.

854768750, 2022-11-23:
#1228

closed 2023-10-02 by srini229