ALIGN-analoglayout/ALIGN-public #1231: Why did router stop choosing the center track?

soneryaldiz, 2022-12-05:
Previously, router was preferring the tracks that are closer to center of two given pins. This behavior has changed recently. See the example below where the right most vertical track is chosen although there are no other blockages. What caused the change in this behavior?

[image]

854768750, 2022-12-05:
I will take a look

854768750, 2022-12-15:
[image]

fixed in https://github.com/ALIGN-analoglayout/ALIGN-public/pull/1236/commits/d39823e38a73c3bf346731f1fe9a8007fb3a5b2d

closed 2022-12-15 by 854768750