ALIGN-analoglayout/ALIGN-public #1255: Do not require power pin removal

soneryaldiz, 2023-01-10:
Power pin removal has been shown to break equivalence between the original and final circuit. There does not exist a layout versus schematic checker that can detect these errors. What does it take to completely eliminate the need for power pin removal?

soneryaldiz, 2023-01-12:
@854768750  and @kkunal1408 , could you please comment on?

@854768750 , could you please scope the effort to refactor place and route so that layout is still generated with these pins and connectivity part of the input design?

854768750, 2023-01-12:
> Power pin removal has been shown to break equivalence between the original and final circuit. There does not exist a layout versus schematic checker that can detect these errors. What does it take to completely eliminate the need for power pin removal?

Is the powerpin removed by PnR?

soneryaldiz, 2023-01-12:
Today, the power and ground ports are completely removed from the hierarchy in the compiler. The question is: if these ports are not removed, can PnR still work?

kkunal1408, 2023-01-12:
@soneryaldiz  Can you explain if you want to completely remove this step or are you only targetting the case where a signal port is connected to power in one instance and a power port in another instance?
The current power port removal was a requirement from PnR as we have a global power grid. For power grid implementation we do not require a port to be created at any hierarchy. As I remember in the schematic we need to add power ports in cases we are not using the common power port definition and most of the analog power is drawn as a separate port. 
In case you want to not delete these ports from subcircuits, how do you want these ports to be created?

soneryaldiz, 2023-01-12:
I am proposing to completely remove this step unless there is a compelling reason or technical challenge. For a given subcircuit, it is straightforward to tell PnR that select pins are power ports and not to be routed as signal. Why remove from the hierarchy?

kkunal1408, 2023-01-12:
Are you suggesting for any sub-hierarchy create a physical power port and then connect it to the grid? What will be the use of these ports?

soneryaldiz, 2023-01-12:
No, not at all. Today, the power ports are removed from the hierarchy for PnR to do its job, right? I am proposing to stop doing this, get rid of power port removal and refactor PnR to still do its job. If this is still not clear, let's talk over a meeting.

kkunal1408, 2023-01-12:
> 

@854768750 Can you answer this question?

854768750, 2023-01-12:
The power pins are removed from signal routing. The power pins are routed in power routing.
I do not understand why the original circuit and final layout are not equivalent.

kkunal1408, 2023-01-27:
In the case, a sub-hierarchy has two types of instantiation one in which a pin is connected to power and another where it is connected to a signal. Removing power pins will create an open in this case.

Soner's idea is to remove this step as deleting a pin causes equivalence not to be straightforward at all levels.

closed 2023-02-07 by soneryaldiz