ALIGN-analoglayout/ALIGN-public #1259: Installation Help

dkaysemi, 2023-01-20:
Sorry for the Basic Question. I installed ALIGN. 
OS: Windows 10 Enterprise
created virtual environment (general)
Running following command as per installation guide. 

Analog_layout\ALIGN\ALIGN-public>python -m cProfile -o stats bin/schematic2layout.py examples/sc_dc_dc_converter

Getting below error.

   from .build_pnr_model import gen_DB_verilog_d
  File "C:\Users\XYZ\Analog_layout\ALIGN\ALIGN-public\align\PnR\build_pnr_model.py", line 14, in <module>
    NType = PnR.NType
AttributeError: partially initialized module 'align.PnR' has no attribute 'NType' (most likely due to a circular import)

Any help to resolve this is appreciated. 

[image]

kkunal1408, 2023-01-26:
Can you provide your C++ and Python versions?

dkaysemi, 2023-01-27:
Hi, Thanks for the response.
C++ = 6.3.0
Python=3.8.8

[image]

kkunal1408, 2023-01-27:
The python and C++ versions look good. The error looks like an incomplete installation. Can you reinstall ALIGN after deleting the _skbuild directory in your ALIGN home area? Let us know if you see any errors during that step.

kkunal1408, 2023-02-03:
The user was able to install it on an Ubuntu machine.

kkunal1408, 2023-02-03:
I tried doing different scenario analyses to reproduce this error on my system. I found that if the GCC path is not set, it does not install the C++-based PnR engine. Please set these before installing for a Linux terminal.

$ export CC=/path/to/your/gcc
$ export CXX=/path/to/your/g++

For csh terminals please use equivalent SETENV commands.

You can check if your C++ installation worked fine or not by running this installation command:
pip install -v -e .[test] --no-build-isolation

Please let us know if you see any errors in this or if this works for you.

PSR0001, 2023-02-06:
Sir, i am trying to install ALIGN on ubuntu(20.04) after 
pip install -v .
gcc version: 9.4.0
python 3.8.10

i got this👇
[image]

kkunal1408, 2023-02-09:
Closing for now due to no response.

closed 2023-02-09 by kkunal1408

hoangtnguyen-3702, 2025-07-02:
> export CXX=/path/to/your/g++

Dear Kunal,

I already tried the two "export" command you recommend, it worked well (did not output anything after these two commands, I suppose in that case there are no errors). However, when I run the command "pip install -v -e .[test] --no-build-isolation", the error is 

  Running command Preparing metadata (pyproject.toml)
  Traceback (most recent call last):
    File "setup.py", line 5, in <module>
      from skbuild import setup
  ModuleNotFoundError: No module named 'skbuild'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "/mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
      main()
    File "/mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
      json_out["return_val"] = hook(**hook_input["kwargs"])
    File "/mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 175, in prepare_metadata_for_build_wheel
      return hook(metadata_directory, config_settings)
    File "/mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/setuptools/build_meta.py", line 174, in prepare_metadata_for_build_wheel
      self.run_setup()
    File "/mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 8, in <module>
      raise AssertionError("Use pip 10+, or install pyproject.toml requirements yourself")
  AssertionError: Use pip 10+, or install pyproject.toml requirements yourself
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /mnt/c/Users/hoangtn/general/bin/python /mnt/c/Users/hoangtn/general/lib/python3.10/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpgru6l8p_
  cwd: /mnt/c/Users/hoangtn/ALIGN_layout_automation/ALIGN-public
  Preparing metadata (pyproject.toml) ... error
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Can you please help me with that?

Thank you very much!