ALIGN-analoglayout/ALIGN-public #1266: [What am I doing wrong?] Assertion `round(HPWL_extend) == round(HPWL_ILP)' failed

rajivbishwokarma, 2023-02-16:
## Complete error generated: 

python3: /home/rb/work/ALIGN-public/PlaceRouteHierFlow/placer/ILP_solver.cpp:3110: double ILP_solver::GenerateValidSolution(const design&, const SeqPair&, const PnRDB::Drc_info&, int): Assertion `round(HPWL_extend) == round(HPWL_ILP)' failed.

## Recreating the error
1. Use the following netlist to run schematic2layout.py.

```
.subckt fn_sim A B C D E F Y vdd GND
M1 net1 A vdd vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M2 net3 B vdd vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M3 net2 C net1 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M4 net2 D net3 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M5 Y E net2 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M6 Y F net2 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=3
M7 Y A net4 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
M8 Y C net4 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
M9 Y E net5 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
M10 net4 B GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
M11 net4 D GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
M12 net5 F GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=3
.ends
```
2. Run 
`schematic2layout.py ../fn_sim -p ../pdks/ALIGN-pdk-sky130/SKY130_PDK`

3. Run time log

```
align.main INFO : Reading netlist: /home/rb/work/ALIGN-public/fn_sim/fn_sim.sp subckt=FN_SIM, flat=0
align.compiler.compiler INFO : Starting topology identification...
align.compiler.compiler INFO : Power and ground nets not found. Power grid will not be constructed.
align.compiler.compiler INFO : Completed topology identification.
align.pnr.main INFO : Running Place & Route for FN_SIM
align.pnr.build_pnr_model INFO : Reading contraint json file FN_SIM.pnr.const.json
align.pnr.build_pnr_model INFO : Reading contraint json file FN_SIM.pnr.const.json
align.pnr.placer INFO : Starting bottom-up placement on FN_SIM 0
PnR.placer.Placer.PlacementCoreAspectRatio_ILP INFO : Required 1 perturbations to generate a feasible solution.
python3: /home/rb/work/ALIGN-public/PlaceRouteHierFlow/placer/ILP_solver.cpp:3110: double ILP_solver::GenerateValidSolution(const design&, const SeqPair&, const PnRDB::Drc_info&, int): Assertion `round(HPWL_extend) == round(HPWL_ILP)' failed.
Aborted (core dumped)
```

4. Output log image
[image]

5. Running schematic2layout.py with m=8 in the input netlist gets stuck in the following step.

`PnR.placer.Placer.PlacementCoreAspectRatio_ILP INFO : Required 1 perturbations to generate a feasible solution. ` 

## Modified netlist:
```
.subckt fn_sim A B C D E F Y vdd GND
M1 net1 A vdd vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M2 net3 B vdd vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M3 net2 C net1 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M4 net2 D net3 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M5 Y E net2 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M6 Y F net2 vdd sky130_fd_pr__pfet_01v8 L=150e-9 W=21e-7 nf=10 m=8
M7 Y A net4 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
M8 Y C net4 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
M9 Y E net5 GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
M10 net4 B GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
M11 net4 D GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
M12 net5 F GND GND sky130_fd_pr__nfet_01v8 L=150e-9 W=10.5e-7 nf=10 m=8
.ends
```
6. Stuck at the following step: 
[image]

rajivbishwokarma, 2023-02-16:
I am not sure what I am doing wrong here? Is there something wrong with the input spice netlist?

kkunal1408, 2023-02-20:
The netlist looks fine. There is issue with the auto constraint generation. I will debug that, in the meantime you can proceed by turning off auto constraints by adding a file "fn_sim.const.json" with content:
```
[
        {"constraint": "PowerPorts", "ports": ["VDD"]},
        {"constraint": "GroundPorts", "ports": ["GND"]},  
        {"constraint": "ConfigureCompiler", "auto_constraint": false}
]
```

You can also add more manual constraints using the options listed here:
https://align-analoglayout.github.io/ALIGN-public/notes/const.html#

kkunal1408, 2023-02-20:
Can you provide a rough schematic or layout plan so that I can debug the auto-generated constraints?

rajivbishwokarma, 2023-02-24:
I solved this issue by connecting the bulk terminals of the transistors to their sources. In the previous schematic, I was connecting the bulk terminals to the VDD for PMOS and to GND for NMOS. 

The following schematic with the bulk terminals of PMOS (C, D, E, F) connected to VDD and the bulk terminal of NMOS (A, C, E) connected to GND was causing the above issue. This circuit fixed it.

[image]

kkunal1408, 2023-02-24:
Thanks a lot. I will check for the error with the previous circuit.

syedimaduddin, 2023-03-20:
I am also facing the same error as @rajivbishwokarma faced with the below schematic of 1bit ADC
[image]

I extracted the netlist from above schematic and converted it to the required format for ALIGN as below
```
.subckt adc_1bit VSS VDD OUT INN INP
XM1 net2 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM2 OUT net2 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM3 net1 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM4 net3 net3 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM5 net1 INN net4 net4 sky130_fd_pr__nfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM6 net2 INP net4 net4 sky130_fd_pr__nfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM7 net4 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM8 OUT net3 VSS VSS sky130_fd_pr__nfet_01v8 L=150e-09 w=10.5e-7 nf=2
XM9 net3 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=150e-09 w=10.5e-7 nf=2
.ends adc_1bit
```
And facing the below error while running ALIGN for sky130 pdk
[image]

kkunal1408, 2023-03-21:
> The netlist looks fine. There is an issue with the auto-constraint generation. I will debug that, in the meantime, you can proceed by turning off auto constraints by adding a file "fn_sim.const.json" with the content:
> 
> ```
> [
>         {"constraint": "PowerPorts", "ports": ["VDD"]},
>         {"constraint": "GroundPorts", "ports": ["GND"]},  
>         {"constraint": "ConfigureCompiler", "auto_constraint": false}
> ]
> ```
> 
> You can also add more manual constraints using the options listed here: https://align-analoglayout.github.io/ALIGN-public/notes/const.html#

@syedimaduddin Does this solution work for you?

syedimaduddin, 2023-03-21:
@kkunal1408 the layout is successfully created using your mentioned method, but the post-layout simulation is different from the pre-layout simulation.

kkunal1408, 2023-03-21:
Are you talking about performance degradation or any specific failures?

syedimaduddin, 2023-03-21:
Actually @kkunal1408, my pre-layout netlist of 1-bit ADC is

```
** sch_path: /home/syedimaduddin/Desktop/VSD_PD_Research_Program/Week-5/xschem/adc_1bit.sch
**.subckt adc_1bit VSS VDD OUT INN INP
*.iopin VSS
*.iopin VDD
*.opin OUT
*.ipin INN
*.ipin INP
XM1 net2 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM2 OUT net2 VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM3 net1 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM4 net3 net3 VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM5 net1 INN net4 net4 sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM6 net2 INP net4 net4 sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM7 net4 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM8 OUT net3 VSS VSS sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
XM9 net3 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 ad='int((nf+1)/2) * W/nf * 0.29' as='int((nf+2)/2) * W/nf * 0.29'
+ pd='2*int((nf+1)/2) * (W/nf + 0.29)' ps='2*int((nf+2)/2) * (W/nf + 0.29)' nrd='0.29 / W' nrs='0.29 / W'
+ sa=0 sb=0 sd=0 mult=1 m=1
V1 VDD GND 1.8
.save i(v1)
V2 VSS GND 0
.save i(v2)
V3 INP GND sin(0.9 0.9 100Meg)
.save i(v3)
V4 INN GND 0.9
.save i(v4)
**** begin user architecture code
.lib /usr/local/share/pdk/sky130A/libs.tech/ngspice/sky130.lib.spice tt
.control
save all
tran 0.01n 100n
plot inp out inn
.endc
**** end user architecture code
**.ends
.GLOBAL GND
.end
```

And my pre-layout simulation using ngspice is
[image]

By converting .spice file to .sp file for ALIGN, my netlist becomes

```
.subckt adc_1bit VSS VDD OUT INN INP
XM1 net2 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-9 W=8.4e-7 nf=4
XM2 OUT net2 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-9 W=8.4e-7 nf=4
XM3 net1 net1 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-9 W=8.4e-7 nf=4
XM4 net3 net3 VDD VDD sky130_fd_pr__pfet_01v8 L=150e-9 W=8.4e-7 nf=4
XM5 net1 INN net4 net4 sky130_fd_pr__nfet_01v8 L=150e-9 W=8.4e-7 nf=4
[... 96 lines cut ...]
C17 m1_312_1484# VSS 0.23fF
C18 m1_1602_1484# INN 0.00fF
C19 m1_1602_1484# m1_312_1484# 0.05fF
C20 m1_312_1484# NMOS_S_74334133_X2_Y1_1679372695_1/a_200_252# 0.22fF
C21 m1_1602_1484# VSS 1.27fF
C22 m1_312_1484# OUT 0.00fF
C23 m1_602_1316# VDD 4.77fF
C24 INP INN 0.04fF
C25 OUT VSS 0.29fF
C26 m1_1602_1484# OUT 0.24fF
C27 INP VSS 0.07fF
C28 VSS 0 0.85fF
C29 OUT 0 1.21fF
C30 VDD 0 12.40fF
C31 m1_1602_1484# 0 4.67fF
C32 m1_312_1484# 0 1.06fF
C33 NMOS_S_74334133_X2_Y1_1679372695_1/a_200_252# 0 1.47fF
C34 NMOS_S_74334133_X2_Y1_1679372695_0/a_200_252# 0 1.47fF
.ends
```

And my post-layout simulation is
[image]

You can see that my pre-layout simulation is not the same as my post-layout simulation for 1-bit ADC. So, I'm saying that it could be caused by error skipping with the constraint you mentioned.

kkunal1408, 2023-03-21:
@syedimaduddin For the schematic simulation, can you try using the updated parameters which you used as input for ALIGN.
e.g.  this is one of your original device parameter
```XM9 net3 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1```
Requirement for ALIGN is that for a device nf should be even. 
This is your modified parameter which you are feeding as input to ALIGN
```XM9 net3 net3 VSS VSS sky130_fd_pr__nfet_01v8 L=150e-9 W=8.4e-7 nf=4```
When you increased your nf to 4 you are changing your circuit device parameters (L modification is only unit change but not sure why you modified your W). While doing your input SPICE  simulations you need to incorporate these changes. Can you please confirm your simulation results after these modifications.

syedimaduddin, 2023-03-22:
@kkunal1408  My problem has been solved after I removed ALIGN from my PC and reinstalled it. An incomplete installation could have caused the error. Maybe I missed some steps when installing ALIGN last time.

This time, I have followed below repositories completely: 
https://github.com/ALIGN-analoglayout/ALIGN-public
https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130

kkunal1408, 2023-03-23:
Thanks for letting us know.

closed 2023-03-23 by kkunal1408