ALIGN-analoglayout/ALIGN-public #1271: Issues while installing ALIGN as a user

Venky767, 2023-03-10:
[image]
I got the above error and then i ran the below command
`> pip install 'networkx>=2.4' 'python-gdsii' 'gdspy' 'pyyaml' 'pybind11' 'pydantic>=1.9.2' 'z3-solver' 'mip' 'more-itertools' 'colorlog' 'plotly' 'numpy' 'pandas' 'werkzeug==2.0.0' 'dash' 'typing_extensions; python_version<"3.8"' 'memory_profiler' 'flatdict' 'pytest' 'pytest-cov' 'pytest-xdist' 'pytest-timeout' 'pytest-cpp'`
Below errors showed up. Please help me debug this.

![Screenshot from 2023-03-11 00-03-34](https://user-images.githubusercontent.com/127543589/224396815-1af0013b-54d7-41b1-a2c3-8bd2
[image]
70efbb6d.png)

Venky767, 2023-03-10:
[image]

kkunal1408, 2023-03-21:
It looks like the issue is with the GDSPY module. Can you try installing it individually?

kkunal1408, 2023-04-07:
CLosing the issue due to no response.

closed 2023-04-07 by kkunal1408