ALIGN-analoglayout/ALIGN-public #1278: ImportError: cannot import name 'PnR' -- on RHEL7

dan-fritchman, 2023-04-14:
I hit this:

```
$ schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/

Traceback (most recent call last):
  File "ALIGN-public/general/bin/schematic2layout.py", line 2, in <module>
    import align
  File "ALIGN-public/align/__init__.py", line 6, in <module>
    from .main import schematic2layout
  File "ALIGN-public/align/main.py", line 14, in <module>
    from .pnr import generate_pnr
  File "ALIGN-public/align/pnr/__init__.py", line 1, in <module>
    from .checkers import *
  File "ALIGN-public/align/pnr/checkers.py", line 7, in <module>
    from .router import NType
  File "ALIGN-public/align/pnr/router.py", line 12, in <module>
    from .. import PnR
ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import) (ALIGN-public/align/__init__.py)
```

With the head of the `master` branch as of this writing (and a few other similar-vintage commits): 

```
$ git log --oneline
c82ffd7e (HEAD -> master, gh/master, gh/HEAD) Merge pull request #1273 from ALIGN-analoglayout/dependabot/npm_and_yarn/dev/Experimental/PlacementEditor/webpack-5.76.1
```

On RedHat Enterprise Linux 7: 

```
$ uname -a
Linux eda-1 3.10.0-1160.83.1.el7.x86_64 #1 SMP Mon Dec 19 10:44:06 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

(I know, RedHat has all these ancient packages, it's a pain, yeah. But it's also the primary (or only) supported OS by a bunch of other EDA software.)

Tried a few different Python setups: 
* The virtualenv recommended in your README, with Python3.8
* Conda environments with Python 3.8, 3.9, 3.10

Installation steps (all from README):

```
# (Configure one of those Python setups listed above )

CC=`which gcc` # 
CXX=`which g++`

python -m pip install pip --upgrade
pip install setuptools wheel pybind11 scikit-build cmake ninja
pip install -v -e .[test] --no-build-isolation
pip install -v --no-build-isolation -e . --no-deps --install-option='-DBUILD_TESTING=ON'
```

FWIW the C/C++ compilers are these versions:

```
$ gcc --version
gcc (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2)

$ g++ --version
g++ (GCC) 9.3.1 20200408 (Red Hat 9.3.1-2)
```

I see there is #1200 with a similar error, and it was resolved. 
That one runs on Ubuntu/ Debian, which also work for me, on a similar vintage of commits.

dan-fritchman, 2023-04-14:
FWIW during those pip-driven build/ installation steps, C++ compilation doesn't appear to be happening at all.

dan-fritchman, 2023-04-17:
I filed https://github.com/scikit-build/scikit-build/issues/939 with scikit-build, and their developer's reply was: 

> We don’t support modern editable installs here, sadley. Initial support is present in scikit-build-core (native builds only ATM). You should also be able to set the envvar for classic editable builds in setuptools.
> I do think we could possibly make this an error though until we can support it. (It is tricky to support because we have to produce a list of file to module mappings for setuptools).

So `skbuild` doesn't claim that these editable installs work at all(?).

srini229, 2023-04-21:
I tried it on an RHEL system with the config:
`Linux bose 3.10.0-1160.59.1.el7.x86_64 #1 SMP Wed Feb 16 12:17:35 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux`
with gcc version `(8.3.0)`. It builds fine in this machine.

Can you try the branch `feature/arm64` using the following steps:
```
git checkout feature/arm64
python -m venv general
source general/bin/activate
pip install --upgrade pip
source install.sh
```

srini229, 2023-04-21:
I have built a wheel for Python 3.8 on an RHEL system with GCC 8.3. The wheel file is in the branch `feature/arm64`. You can try installing the wheel directly on your system using the following commands:
```
git pull
git checkout feature/arm64
python -m venv general
source general/bin/activate
pip install --upgrade pip
pip install dist/align-0.9.8-cp38-cp38-linux_x86_64.whl
```

srini229, 2023-05-12:
Please reopen if the issue persists.

soneryaldiz, 2023-05-15:
@dan-fritchman , I am experiencing the same error and filed #1284 . Please try without upgrading `pip`.

dan-fritchman, 2023-05-19:
Terrific thanks! 

For reference, this version of pip: 
```
$ pip --version
pip 21.2.4
```
worked for me.

closed 2023-05-20 by srini229