ALIGN-analoglayout/ALIGN-public #1279: Opens and DRC violations in ALIGN generated layout

rahulearn2019, 2023-04-15:
The circuit in xschem
[image]

The netlist generated by xschem

```
** sch_path: /home/rahul/Documents/logicLine/logicLine.sch
.subckt logicLine RBLprecharge RBLprechargeEnable VDD GND OR NOR
*.PININFO RBLprecharge:I RBLprechargeEnable:I VDD:B GND:B OR:O NOR:O
XC1 net1 GND sky130_fd_pr__cap_mim_m3_1 W=1 L=1 MF=1 m=1
XM1 OR net1 GND GND sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 m=1
XM3 NOR OR VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=2 nf=1 m=1
XM4 NOR OR GND GND sky130_fd_pr__nfet_01v8 L=0.15 W=1 nf=1 m=1
XM2 OR net1 VDD VDD sky130_fd_pr__pfet_01v8 L=0.15 W=2 nf=1 m=1
XM5 net1 RBLprechargeEnable RBLprecharge VDD sky130_fd_pr__pfet_01v8 L=0.15 W=2 nf=1 m=1
.ends
.end
```

Modified netlist given to ALIGN

```
.subckt logicLine RBLprecharge RBLprechargeEnable VDD GND OR NOR
C1 net1 GND sky130_fd_pr__cap_mim_m3_1 W=1000e-9 L=1000e-9
XM1 OR net1 GND GND sky130_fd_pr__nfet_01v8 L=150n W=420n nf=2
XM3 NOR OR VDD VDD sky130_fd_pr__pfet_01v8 L=150n W=840n nf=2
XM4 NOR OR GND GND sky130_fd_pr__nfet_01v8 L=150n W=420n nf=2
XM2 OR net1 VDD VDD sky130_fd_pr__pfet_01v8 L=150n W=840n nf=2
XM5 net1 RBLprechargeEnable RBLprecharge VDD sky130_fd_pr__pfet_01v8 L=150n W=840n nf=2
.ends
```

Errors - It says there are opens in the design when I cannot see any. There are some DRC violations which can be dealt with in magic
[image]

[image]

rahulearn2019, 2023-04-15:
When I open the layout in magic, it doesn't show any DRC though
[image]

srini229, 2023-07-26:
@rahulearn2019 The reported error is an OPEN. It is an LVS error. It is most likely occurring at the contacts of the MIM capacitor which the ALIGN internal checker might erroneously be reading as OPEN. If you find there are no LVS errors in the layout, you can ignore the ALIGN-reported error. We'll fix the internal checker.