ALIGN-analoglayout/ALIGN-public #1280: Unable to setup ALIGN

xzores, 2023-05-02:
I get the error `ERROR: Could not build wheels for align, which is required to install pyproject.toml-based projects`. 
I cannot find a solution

srini229, 2023-05-11:
Can you provide the commands you used and your installation environment such as python/gcc version?

xzores, 2023-05-13:
I went from a clean install got a bunch of errors and downloaded the newest version of python, GCC, cmake, ninja and wheels (i think) that seemed to get me further. Everything is up to date.

Now I am stuck at this error 

Also the installation is very very very slow taking something like 20 min before failing on a fast PC.

```
  An error occurred while building with CMake.
    Command:
      /tmp/pip-build-env-tuizgatd/overlay/lib/python3.8/site-packages/cmake/data/bin/cmake --build . --target install --config Release --
    Install target:
      install
    Source directory:
      /mnt/d/GitHub/Other stuff/ALIGN-public
    Working directory:
      /mnt/d/GitHub/Other stuff/ALIGN-public/_skbuild/linux-x86_64-3.8/cmake-build
  Please check the install target is valid and see CMake's output for more information.

  error: subprocess-exited-with-error

  × Building wheel for align (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  full command: /usr/bin/python3 /home/xzores/.local/lib/python3.8/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py build_wheel /tmp/tmpng1aq4xw
  cwd: /mnt/d/GitHub/Other stuff/ALIGN-public
  Building wheel for align (pyproject.toml) ... error
  ERROR: Failed building wheel for align
Failed to build align
ERROR: Could not build wheels for align, which is required to install pyproject.toml-based projects
```

srini229, 2023-06-27:
It appears like you are installing on a WSL system.
If so, can you try to install it in a directory within the Linux fs and not inside the Windows drive?
If you are not using a WSL system, can you redirect the pip install commands output to a log file and attach the log to this issue to help debug?

xzores, 2023-06-28:
My PC is down atm, I will give it another try when I get a new one ;]

soneryaldiz, 2023-07-07:
@srini229 , install as a developer option no longer seems to work. _skbuild is no longer created and C++ code does not seem to be compiled.

soneryaldiz, 2023-07-07:
```bash
DEPRECATION: --build-option and --global-option are deprecated. pip 23.3 will enforce this behaviour change. A possible replacement is to use --config-settings. Discussion can be found at https://github.com/pypa/pip/issues/11859
WARNING: Implying --no-binary=:all: due to the presence of --build-option / --global-option. 
```

soneryaldiz, 2023-07-07:
Workaround is to use older pip version: `pip install "pip==20.*"`

srini229, 2023-07-08:
@soneryaldiz Does the use of :
`pip install -v --no-build-isolation -e . --no-deps --config-settings='--build-type=Release' --config-settings='-DBUILD_TESTING=ON'`
help?

soneryaldiz, 2023-07-10:
@srini229 , I think this issue is resolved by the latest PR. I am able to install and test on SLES12.

srini229, 2023-07-24:
Please reopen if you see the issue persists.

closed 2023-07-24 by srini229