ALIGN-analoglayout/ALIGN-public #1281: "schematic2layout.py: Command not found." on centos linux

PLJSNPS, 2023-05-08:
I am trying to install align on CentOS Linux but getting the the following error:
"schematic2layout.py: Command not found."
Here are the steps I am using. The dark parts are modified after I tried several times.

Anybody can help me to figure out what I am missing? Thanks a lot

1. OS:
 NAME="CentOS Linux" ,VERSION="7 (Core)"
2. C++: 
CC=/depot/gcc-6.2.0/bin/gcc
CXX=/depot/gcc-6.2.0/bin/g++
3. Python:
Python 3.8.3
5. Steps: 
   1) git clone https://github.com/ALIGN-analoglayout/ALIGN-public
   2) cd ALIGN-public
   3) python -m venv general
   4) source general/bin/**activate.csh**
   5) python -m pip install pip --upgrade
   6) **pip install --upgrade pip setuptools (wheel)**
   7) pip install -e **‘.[test]‘**
   8) pip install setuptools wheel pybind11 numpy scikit-build cmake ninja
   9) pip install -v -e **‘.[test]‘** --no-build-isolation
 10) pip install -v --no-build-isolation -e . --no-deps **-install-option**='--build-type=Release' **-install-option**='-DBUILD_TESTING=ON'
 11) mkdir work && cd work
 12) schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/

srini229, 2023-05-11:
Does the installation complete without any errors?
If so, can you try the command `rehash` if you are using a `csh` or its derivatives?

srini229, 2023-05-22:
Can you pull the latest master and try? the `--install-option` is no longer supported in pip. Replace it with `--global-option`.

srini229, 2023-06-14:
No update. Closing the issue.

closed 2023-06-14 by srini229