ALIGN-analoglayout/ALIGN-public #1292: Align Installation Error

DolanKeelan, 2023-06-14:
I'm facing an issue when trying to install the ALIGN tool. I've also attached the image of the error below. Thanks for any help offered.
Error: 
subprocess-exited-with-error
  
  × Building wheel for align (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

Python Version: 3.9.7
GCC Version: 9.3.0
Pip version: 23.1.2

Commands:
- cd ALIGN-public
- python -m venv general
- source general/bin/activate.csh
- python -m pip install pip --upgrade
- pip install -v .

[image]

srini229, 2023-06-14:
You mentioned the python version you are using is `3.9.7`. The install uses version `3.7` in the log image you shared. Can you please check the version of python used to create the virtual environment?

DolanKeelan, 2023-06-14:
It was my mistake. My gcc and g++ variables were not set up correctly and thus getting these pointed to the correct directories fixed the problem. All is installed correctly now :) Thanks for the help!