ALIGN-analoglayout/ALIGN-public #1294: Place Using Python

mahfuzalhasan, 2023-06-27:
Hello,
Thank you for this outstanding task. I am currently working on top of this repo. I had 2 queries.

**1.** Can this repo be modified for the layout generation of traditional MOSFET technologies? 
**2.** I was just wondering what I can do to utilize the placer_pythonic.py. 
The code calls placer_driver() from pnr/main.py and then eventually it goes into C++ modules. I want to accomplish the whole place and routing task using python. It seems like you have put the placer functions for python in pythonic_placer.py and pythonic_placer_sp.py. Can you please guide me how should I process the input to utilize these pythonic placer functions?

arvuce22, 2023-06-27:
Hi,
Regarding your 1st point: We have used the same framework to implement SkyWater 130 pdk, which you can follow from the ALIGN sub-repo: https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130
There are also mock Bulk 65 and 40nm PDKs developed based on literature, which you can find in the pdks folder of this repository.

We will get back to you in sometime for your 2nd query. 

Thanks!

srini229, 2023-06-27:
Hi,

For your second query:
The Pythonic placers are still under development and do not have all the features of the C++ placer. They are also not accessible from the `schematic2layout.py`. You are welcome to explore the placers using the syntax in the files:`tests/placer_pythonic/test_placer_pythonic.py`

srini229, 2023-07-24:
Please re-open if you have further queries.

closed 2023-07-24 by srini229

giuseppe-chiari, 2025-02-24:
Are there any updates on this matter?