ALIGN-analoglayout/ALIGN-public #1295: Unble to have ALIGN setup correctly

ChrisZonghaoLi, 2023-07-04:
I have align-0.9.8 installed successfully. However, whenever I run `schematic2layout -h` it pops up this following error:

[image]

It looks like it is complaining about some stuff that is dependent on `pydantic`. Any idea please? 

P.S. I had a similar issue with ALIGH after the installation that was complaining about some functions of `pedantic`, which were removed in the new V2 version. I downgraded it to v1.10 and the old complaint was gone.

ChrisZonghaoLi, 2023-07-05:
Okay, I managed to solve the problem. I actually found that despite my system meeting all your needed requirements in `README`, I did the following few things to let it work:

1. Install the latest version `cxx-compiler` (https://stackoverflow.com/questions/69485181/how-to-install-g-on-conda-under-linux): conda install -c conda-forge cxx-compiler
2. Install latest `gcc`:  conda install -c conda-forge gcc
3. Downgrade `pydantic` to V1.10: pip installl pydantic==1.10

These worked for me.

closed 2023-07-05 by ChrisZonghaoLi

srini229, 2023-07-09:
Thank you for the update.