ALIGN-analoglayout/ALIGN-public #1296: Differeent transistor layout for SKY130 process

ChrisZonghaoLi, 2023-07-06:
So I spent quite some time looking into the layout generated by ALIGN in SKY130 process, and it seems that it can only generate `RVT` transistor layout (which all later interpreted as `sky130_fd_pr__nfet_01v8` after the layout extraction), despite even the transistor in the netlist is not `RVT`. Also, I think it is lacking the options such as high-voltage transistors. 

I guess I need to implement these transistor layouts myself then. However, I had a hard time really finding useful information on how to do this (like a tutorial) as the documentation misses lots of details. Could everyone here show me briefly how to do so?

Thanks.

ChrisZonghaoLi, 2023-07-06:
I also had a hard time locating where you define if a transistor is belonging to which `vt_type`.

kkunal1408, 2023-07-06:
Hi Chris,
Thanks for your feedback. Yes, we have limited features in the SKY130 process and we are open to collaboration.
To add any new VT type you can use these steps (this es example from mock-pdk but it can be extended to SKY130):

1. Add a new cell type in models.sp and map that to a mos in [models.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/FinFET14nm_Mock_PDK/models.sp). This will help in identifying the device as nmos/pmos device for any compiler optimization.
2. Add a different "vt_type" in [layers.json](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/FinFET14nm_Mock_PDK/layers.json) . 
3. Create cell generator param to add vt_type by string-based mapping from the actual name of the device in [gen_param.py](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/FinFET14nm_Mock_PDK/gen_param.py) .
4. Add a VT layer in MOS generator in [mos.py](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/FinFET14nm_Mock_PDK/mos.py)

Please let us know if it works for you. 
@arvuce22 Let me know if you have more insight about SKY130 PPDK.

ChrisZonghaoLi, 2023-07-06:
Thanks @kkunal1408 for the prompt feedback. Yes, I would like to expand the layout library in SKY130 PDK (and possibly GF180MCU as well in the future). 

The first step should be simple as it is just a model name, which is used to lookup the netlist for the right device.

I guess the difficult part is to generate the corresponding layout for different transistor devices. Again, I assume the current ability is to only draw the layout for `sky130_fd_pr__nfet_01v8`.

I have also looked into `gen_param.py` which I believe is where the spice2layout process happens (correct me if I am mistaken). Now, after inserting some `print` commands in the `gen_param.py` to help me understand what is going on under the hood, I have run the example `buffer` in the `ALIGN-pdk-sky130/examples` and the following figure contains some questions that I like to ask (to help me also implement other layouts):

[image]

I kinda see that it instantiates two subcircuits: `NMOS_S_29520189` and `PMOS_S_61174647`. I would like to know where to define these two primitives. For their other attributes, such as `vt_type`, I did not find an explicit location to define them. Perhaps @arvuce22 would be the best person to ask this question?

Thanks.

kkunal1408, 2023-07-08:
Gen_param.py is not that difficult to understand: Here we are converting the device parameters to arguments for most generator. As you can see in the figure below the highlighted keyword will be used to identify the vt_type of the transistor.
[image]

After this in the layers.json you can see these attributes which define the vt type allowed for the PDK:
"vt_type":["HVT", "LVT", "RVT"],

I see that the mos.py already has the LVT and HVT layers:
[image]

I am not sure about the actual layout of the transistor, but I checked now and we already have HVT, LVT, and RVT devices for SKY130PDK. The buffer example in the PDK directory has dummy example devices of nmos_rvt and pmos_rvt and these are not actual PDK-defined devices. Can you provide the device names for LVT and HVT devices you are using? Based on that I can help you with the first 3 steps. For the last step, I would need more info on how to add the LVT layer to the layout of the transistor.

ChrisZonghaoLi, 2023-07-08:
> Gen_param.py is not that difficult to understand: Here we are converting the device parameters to arguments for most generator. As you can see in the figure below the highlighted keyword will be [...]

Hi @kkunal1408 thanks for the reply. What is also making me wonder is exactly what you have mentioned here: the `nmos_rvt` and `pmos_rvt` are not the actual PDK-defined devices. However, when I am running ALIGN and use Magic to view the layout, ALIGN is still able to generate a DRC-free buffer layout, despite I am not defining any PDK-define transistor model in this buffer netlist example (I understand it is defined in `models.sp` but is it enough? However, it is not a real transistor in SKY130 at all. In contrast to `five_transistor_ota` where it has `sky130_fd_pr__nfet_01v8` defined as the transistor being used). Also, after the Magic post layout extraction,  `nmos_rvt` has been recognized as `sky130_fd_pr__nfet_01v8`. Therefore I am a bit lost here regarding how this netlist2layout translation work here... In addition, how could I define the VT type for a transistor?

What I was thinking is, there should be somewhere where I can do this to define a library of transistor layouts:
'define a transistor model for a PDK' -> 'define its VT type (if really necessary)' -> 'define its layout'
or something similar to this.

Regarding the transistor I am using, I am not sure if you can strictly just use these three VT (LVT, RVT, HVT) types to really classify it. I am using `sky130_fd_pr__nfet_g5v0d10v5` and `sky130_fd_pr__pfet_g5v0d10v5` for some LDO applications, their nominal VT is around 0.8-1V. You can see more details of it from here:https://github.com/google/skywater-pdk/blob/main/docs/rules/device-details.rst

Thanks a lot!

Chris

kkunal1408, 2023-07-08:
It is a mapping method from a name in the netlist to an actual layout. 
As you can see in the models.sp, we can map multiple names to the same default transistor nmos. The default nmos is mapped to generate a layout of sky130_fd_pr__nfet_01v8. 

```
.model sky130_fd_pr__nfet_01v8 nmos w=1 l=1
.model nmos_rvt nmos l=1 w=1 nfin=1 nf=1 m=1  stack=1 parallel=1
.model nfet nmos nfin=1 nf=1 l=1 m=1  stack=1 parallel=1
```
It means, if you add ```.model sky130_fd_pr__nfet_g5v0d10v5 nmos w=1 l=1 ```  it will map sky130_fd_pr__nfet_g5v0d10v5  first to default nmos and thus generate sky130_fd_pr__nfet_01v8. 
Now to generate the correct sky130_fd_pr__nfet_g5v0d10v5 , we need to identify first how this new transistor is different from the default transistor sky130_fd_pr__nfet_01v8:  does it need an extra LVT layer or a different dimension of any specific layer? These layer addition/modifications you need to create in mos.py. After that to map the creation of that extra layer or dimension modification to the name sky130_fd_pr__nfet_g5v0d10v5 happens in gen_param.py and then passed as an argument to mos.py.

ChrisZonghaoLi, 2023-07-10:
Thanks @kkunal1408, and I also had a good meeting with @arvuce22 today; he explained the flow to me briefly, which was very helpful. I also get what you are saying and will investigate what layers `sky130_fd_pr__nfet_g5v0d10v5` contains. But maybe my last question, which @arvuce22 also would like me to ask you is how do you define the `vt_type` for a transistor? For example, what if I want to define `sky130_fd_pr__nfet_g5v0d10v5` as an `HVT` device, how should I do so?

kkunal1408, 2023-07-28:
Hi Chris,
If you look into the gen_param.py, it is a simple string based match right now.
```
if "vt_type" in design_config:
            vt = [vt.upper() for vt in design_config["vt_type"] if vt.upper() in subckt.elements[0].model]

```
Here,
subckt.elements[0].model will contain your  actual name of the device ```sky130_fd_pr__nfet_g5v0d10v5```
We assume all devices in a primitive to be of the same VT_type, so we are checking only for the first device.

design_config["vt_type"] contains different VT_types keywords defined in layers.json

You can modify this line create any kind of mapping.

srini229, 2023-10-02:
@ChrisZonghaoLi Is there anything pending to be done in this issue?

arvuce22, 2023-10-02:
@srini229 We have almost resolved the issues. 
@ChrisZonghaoLi If you are having any issues, please feel free to create new issues in ALIGN SKy130 repo: https://github.com/ALIGN-analoglayout/ALIGN-pdk-sky130

closed 2023-10-02 by arvuce22