ALIGN-analoglayout/ALIGN-public #1299: Logic Behind Device Parameter Calculation.

mahfuzalhasan, 2023-07-10:
Hello,
I was looking into the code for MOS generator for **Sky 130** and I am little bit confused about some of the logic in the implementation. It will be really great if you kindly respond to the following queries, please.

1. The **_height_** that's passed to the MOSGenerator Class is set to 28 by default. I was just wondering what the reason for it. And also, what does it mean by height there?
[image]

2. The height is assigned to finsPerUnitCell variable in the MOSGenerator class instantiation. Later using **finsPerUnitCell** and **fin**, variable **finDummy** is calculated and a constraint is set (finDummy >= 8). What's the purpose of this finDummy as it hasn't been used elsewhere. Also, in this stage during gate calculation it always gets multiplied by two (marked in the attached image). I couldn't find any reason for that.
[image]

3. In the finDummy calculation, variable **_fin_** is used. Bit **_fin_** is calculated in the gen_param.py, using width and Fin Pitch. 
                **size = int(float(mvalues[key]["W"])*1E+9/design_config["Fin_pitch"])**

But later in mos.py the above calculated size has been used as **_fin_**. And this **_fin_** is used to calculate the active width.
                 **_self.activeWidth =  self.pdk['Fin']['Pitch']*fin_**
[image]

       So, what's the significant of this fin if only it's used in the reverse calculation? 
Also **_fin_** is calculated using width and later is used to calculate finDummy with height.  I couldn't comprehend why these variables are related?

The reason I am asking all of those implementation-oriented question is because if I want to add my own device how I should modify these parameters associated with above questions.

Thank you for the implementation. I am looking forward to hearing from you.

arvuce22, 2023-07-13:
Following is a response to your questions:
The "height" parameter defines a row height (i.e., active width + active to active spacing). By default, we are using it equal to 28 (i.e., the row height will be 28*Fin["Pitch"]). The "height" consists of two parts i.e., i) width of the device which defines "fin" and ii) findummy (which defines minimum active to active spacing in terms of Fin["Pitch"]).

For example:  If Fin[Pitch] = 100nm and height = 28 and Findummy = 4 -> You can have a device with maximum width = 28*100nm - 2*4*100nm (4 is for Findummy and 2 to consider above and below active to active spacing). Therefore, Max. Width = 2000nm or 2um, if you have a device in your netlist with W > 2um, it will result in an error because Findummy will be < 4 (that violates active to active spacing rule). You can change default height (keep it multiple of 4, e.g., 24 or 20, etc.). Make sure "height" is selected such that it can work with each device width in your netlist (It will be same for all devices in a circuit, you cannot change it individually). 

We will update the code in the coming weeks to make it simple and clear.

srini229, 2023-10-02:
@arvuce22 can we close this issue and track he comments in a separate issue in the SKY130 repo?

arvuce22, 2023-10-02:
@srini229 Sure!

closed 2023-10-02 by arvuce22