ALIGN-analoglayout/ALIGN-public #1302: Code error

breeze12380, 2023-08-15:
Installed ALIGN successfully. But when I run 'schematic2layout.py -h', it gives error:

Traceback (most recent call last):
  File "/general/bin/schematic2layout.py", line 2, in <module>
    import align
  File "/ALIGN-public/align/__init__.py", line 6, in <module>
    from .main import schematic2layout
  File "/ALIGN-public/align/main.py", line 11, in <module>
    from .compiler import generate_hierarchy
  File "/ALIGN-public/align/compiler/__init__.py", line 1, in <module>
    from .compiler import generate_hierarchy, read_lib
  File "/ALIGN-public/align/compiler/compiler.py", line 4, in <module>
    from ..schema import SubCircuit, constraint
  File "/ALIGN-public/align/schema/__init__.py", line 3, in <module>
    from .subcircuit import SubCircuit, Circuit
  File "/ALIGN-public/align/schema/subcircuit.py", line 8, in <module>
    from .constraint import ConstraintDB
  File "/ALIGN-public/align/schema/constraint.py", line 758
    if value := getattr(cls, attr, False):
              ^
SyntaxError: invalid syntax

srini229, 2023-08-16:
Can you provide the version of Python and the platform you are trying it on?

breeze12380, 2023-08-16:
Python 3.7.11. I tried two machines, each with and without virtual env. One attempt yields the error above. Wheel generation failed for other attempts, which seems not uncommon based on other threads. I installed and ran the flow with the example successfully around a year ago. What changes were made? If not, I think the issue results from my updated python version.

srini229, 2023-08-17:
walrus operator (:=) was introduced in Python 3.8. Can you try on Python 3.8 or a later version? 
We'll update the documentation to reflect this.