ALIGN-analoglayout/ALIGN-public #1315: ImportError: cannot import name 'PnR'

lal11s, 2023-11-18:
I am getting the problem when I try to run the align command "schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/ --viewer".I followed the installation guide but still got this problem.
$ schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/ --viewer
Traceback (most recent call last):
  File "/home/liaolelele/ALIGN-public/general/bin/schematic2layout.py", line 2, in <module>
    import align
  File "/home/liaolelele/ALIGN-public/align/__init__.py", line 6, in <module>
    from .main import schematic2layout
  File "/home/liaolelele/ALIGN-public/align/main.py", line 14, in <module>
    from .pnr import generate_pnr
  File "/home/liaolelele/ALIGN-public/align/pnr/__init__.py", line 1, in <module>
    from .checkers import *
  File "/home/liaolelele/ALIGN-public/align/pnr/checkers.py", line 7, in <module>
    from .router import NType
  File "/home/liaolelele/ALIGN-public/align/pnr/router.py", line 12, in <module>
    from .. import PnR
ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import) (/home/liaolelele/ALIGN-public/align/__init__.py)

srini229, 2023-11-18:
Did you install ALIGN using the instructions in the [https://github.com/ALIGN-analoglayout/ALIGN-public](https://github.com/ALIGN-analoglayout/ALIGN-public)? If so, can you rename the align directory to _align and try running the example?

lal11s, 2023-11-19:
@srini229Thank you for your support  I did what you said，but it didn't work.
` File "/home/liaolelele/ALIGN-public/general/bin/schematic2layout.py", line 2, in <module>
    import align
ModuleNotFoundError: No module named 'align'
(general) liaolelele@liaolelele-virtual-machine:~/ALIGN-public/work$ schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/ --viewer
Traceback (most recent call last):
  File "/home/liaolelele/ALIGN-public/general/bin/schematic2layout.py", line 2, in <module>
    import align
  File "/home/liaolelele/ALIGN-public/align/__init__.py", line 6, in <module>
    from .main import schematic2layout
  File "/home/liaolelele/ALIGN-public/align/main.py", line 14, in <module>
    from .pnr import generate_pnr
  File "/home/liaolelele/ALIGN-public/align/pnr/__init__.py", line 1, in <module>
    from .checkers import *
  File "/home/liaolelele/ALIGN-public/align/pnr/checkers.py", line 7, in <module>
    from .router import NType
  File "/home/liaolelele/ALIGN-public/align/pnr/router.py", line 12, in <module>
    from .. import PnR
ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import) (/home/liaolelele/ALIGN-public/align/__init__.py)
`

kkunal1408, 2023-11-19:
From the error, it looks like there was an issue installing the PnR module which is in C++. Did you observe any errors regarding that during installation?

lal11s, 2023-11-19:
there is no error during the installation,and my partner met the same problem.

kkunal1408, 2023-11-19:
Can you share your installation log?

lal11s, 2023-11-19:
@kkunal1408 sorry,I don't know how to check and store my installation log in Ubuntu.Could you please tell me how to do it?

kkunal1408, 2023-11-19:
You can just redirect the output to a file install-align-command > installation.log

srini229, 2023-11-20:
You can try the docker image with ALIGN from [https://hub.docker.com/r/darpaalign/align-public](https://hub.docker.com/r/darpaalign/align-public). This does not require any ALIGN installation; just a working docker setup.

srini229, 2023-11-28:
Please reopen if the suggestions didn't help.