ALIGN-analoglayout/ALIGN-public #1321: Docker image build failing on pip install

Mario1159, 2024-02-28:
Building the image fails on pip install building the ILP Solver Interface without printing a detailed error.
I am currently building the image like this on `master` branch:
```sh
ALIGN-public> docker build -t align -f .\docker\dockerfile .
```
Also tried doing pip install locally (outside the docker image) without luck.
Building the ILP Solver Interface manually from its own repository works ok.
```sh
ILPSolverInterface> cmake -B build .
ILPSolverInterface> cmake --build build
```
Log of the error section, this should be reproducible building the image though not sure why the workflow build pass.
```md
284.0   configure: In case of trouble, first consult the troubleshooting page at https://projects.coin-or.org/BuildTools/wiki/user-troubleshooting
284.0   configure: Configuration of Cbc successful
284.0   configure: In case of trouble, first consult the troubleshooting page at https://projects.coin-or.org/BuildTools/wiki/user-troubleshooting
284.0   configure: Main configuration of Cbc successful
284.0   ninja: build stopped: subcommand failed.
284.0
284.0   Traceback (most recent call last):
284.0     File "/tmp/pip-build-env-4s9zh0l6/overlay/local/lib/python3.10/dist-packages/skbuild/setuptools_wrap.py", line 674, in setup
284.0       cmkr.make(make_args, install_target=cmake_install_target, env=env)
284.0     File "/tmp/pip-build-env-4s9zh0l6/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 697, in make
284.0       self.make_impl(clargs=clargs, config=config, source_dir=source_dir, install_target=install_target, env=env)
284.0     File "/tmp/pip-build-env-4s9zh0l6/overlay/local/lib/python3.10/dist-packages/skbuild/cmaker.py", line 742, in make_impl
284.0       raise SKBuildError(msg)
284.0
284.1   An error occurred while building with CMake.
284.1     Command:
284.1       /tmp/pip-build-env-4s9zh0l6/overlay/local/lib/python3.10/dist-packages/cmake/data/bin/cmake --build . --target install --config Release --
284.1     Install target:
284.1       install
284.1     Source directory:
284.1       /ALIGN-public
284.1     Working directory:
284.1       /ALIGN-public/_skbuild/linux-x86_64-3.10/cmake-build
284.1   Please check the install target is valid and see CMake's output for more information.
284.1
285.0   error: subprocess-exited-with-error
285.0
285.0   × Building wheel for align (pyproject.toml) did not run successfully.
285.0   │ exit code: 1
285.0   ╰─> See above for output.
```

srini229, 2024-02-29:
I will try to reproduce this error. Is there any particular reason you are building the image and not using the one on docker hub?

srini229, 2024-02-29:
I was able to build the image fine on my system running Debian 12:
```
 => [internal] load build definition from dockerfile                                                                                                                                             0.0s
 => => transferring dockerfile: 1.56kB                                                                                                                                                           0.0s
 => [internal] load metadata for docker.io/library/ubuntu:22.04                                                                                                                                  0.0s
 => [internal] load .dockerignore                                                                                                                                                                0.0s
 => => transferring context: 2B                                                                                                                                                                  0.0s
 => [ 1/17] FROM docker.io/library/ubuntu:22.04                                                                                                                                                  0.0s
 => [ 2/17] RUN apt update -y                                                                                                                                                                    7.4s
 => [ 3/17] RUN apt install -y git python3 python3-pip gcc g++ lp-solve sudo liblpsolve55-dev                                                                                                   41.7s
 => [ 4/17] RUN apt clean                                                                                                                                                                        0.3s
 => [ 5/17] RUN git clone https://github.com/ALIGN-analoglayout/ALIGN-public.git                                                                                                                22.0s
 => [ 6/17] WORKDIR /ALIGN-public                                                                                                                                                                0.0s
 => [ 7/17] RUN mkdir -p /data/wheels                                                                                                                                                            0.4s
 => [ 8/17] RUN pip download setuptools wheel pybind11 scikit-build cmake ninja networkx python-gdsii gdspy pyyaml pybind11 'pydantic>=1.9.2,<=1.20' z3-solver mip more-itertools colorlog plo  12.6s
 => [ 9/17] RUN pip install --find-links=/data/wheels --no-index --upgrade pip                                                                                                                   1.1s
 => [10/17] RUN pip install --find-links=/data/wheels --no-index setuptools wheel pybind11 scikit-build cmake ninja networkx python-gdsii gdspy pyyaml pybind11 'pydantic>=1.9.2,<=1.20' z3-so  12.3s
 => [11/17] RUN env BUILD_TYPE='Release' pip install -v . --no-deps                                                                                                                            506.4s 
 => [12/17] RUN echo '#!/bin/bash\nif [[ -d "${ALIGN_PDK_DIR}" ]] && [[ -d "${ALIGN_CKT_DIR}" ]]; then\n\techo "PDK directory : ${ALIGN_PDK_DIR}"\n\techo "Netlist directory : ${ALIGN_CKT_DIR}  0.2s 
 => [13/17] RUN chmod +x /usr/local/bin/run_align                                                                                                                                                0.4s 
 => [14/17] RUN echo "export OPENBLAS_NUM_THREADS=1" >> /etc/bash.bashrc                                                                                                                         0.4s 
 => [15/17] RUN rm -rf /data/wheels /ALIGN-public/_skbuild                                                                                                                                       1.5s 
 => [16/17] WORKDIR /work                                                                                                                                                                        0.0s 
 => exporting to image
```

Can you let me know the platform you are trying to build the image on?

Mario1159, 2024-02-29:
That is crazy strange, it indeed works building on linux, but not in windows.
The reason I was trying to build the docker image was just to test in a contained environment the `pip install` that wasn't working. I am currently trying to call the `schematic2layout` tool directly from python instead of using the docker image.
I don't know very well how docker works in the back, but if I had to take a guess maybe the problem was trying to link the linux precompiled libraries from the ILP interface in a docker container executed from windows.
Knowing that works in linux solves the issue for me by now, but maybe I can try later to do a install with a compiled from source version of the interface.

srini229, 2024-02-29:
I will check the docker build on Windows and get back. Also, I plan to simplify the ILP solver interface by pre-building it. That will simplify the build process eventually.