ALIGN-analoglayout/ALIGN-public #1331: About bulk pdks, I dont have the fin layer in my pdk.

Nakiana, 2024-06-21:
Hello. Regarding bulk PDKs, my PDK does not include the fin layer. Additionally, I do not utilize the process parameters in the .sp file `nfin=8 nf=4`. How should I proceed with using ALIGN?

Furthermore, I noticed the sentence "We will remove the 'Fin' layer for bulk PDKs in future releases" in the PDK Abstraction Guide.pdf. Does this mean ALIGN layout generator can only be used with finFET processes for now?
[image]

srini229, 2024-06-21:
ALIGN supports bulk PDKs with the `Fin` layer used as a place holder for certain assumptions in the MOS layout generator.  Example mock pdks in the `pdks` directory: `Bulk65nm_Mock_PDK` and `Bulk45nm_Mock_PDK`

Nakiana, 2024-06-21:
> ALIGN supports bulk PDKs with the `Fin` layer used as a place holder for certain assumptions in the MOS layout generator. Example mock pdks in the `pdks` directory: `Bulk65nm_Mock_PDK` and `Bulk45nm_Mock_PDK`

Could you please clarify ALIGN how to distinguish between the `Fin `layer when used as a placeholder and when it functions as a useful layer in FinFET technology? Additionally, if I'm utilizing the `Bulk45nm_Mock_PDK`, what should the parameters `nfin=8 nf=4` be set to? Are there any specific rules for setting these parameters?
[image]

Nakiana, 2024-06-25:
@srini229 Could you please clarify ALIGN how to distinguish between the `Fin `layer when used as a placeholder and when it functions as a useful layer in FinFET technology? Additionally, if I'm utilizing the `Bulk45nm_Mock_PDK`, what should the parameters `nfin=8 nf=4` be set to? Are there any specific rules for setting these parameters?

kkunal1408, 2024-06-25:
The update from the slides have already been included in the latest version of code. For ```Bulk45nm_Mock_PDK``` ALIGN just uses ```W, NF (number of fingers), and M``` parameters. Here is an example of SPICE netlist in Bulk technology
[five_transistor_ota_Bulk.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/examples/five_transistor_ota_Bulk/five_transistor_ota_Bulk.sp)

You can look at this code to understand the handling of these parameters ([gen_param.py](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/Bulk45nm_Mock_PDK/gen_param.py)).

Nakiana, 2024-06-25:
@kkunal1408 Oh, I think I understand now. Thank you very much!

closed 2024-06-25 by Nakiana