ALIGN-analoglayout/ALIGN-public #1333: align.cell_fabric.drc ERROR : Found errors: DRC 2

Nakiana, 2024-06-25:
Hello, I only edited my `layers.json` file and left the Python files unchanged. I used [five_transistor_ota_Bulk.sp](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/examples/five_transistor_ota_Bulk/five_transistor_ota_Bulk.sp) for testing, but encountered a `DRC` error. Below is the log:
```
2024-06-25 15:21:34 align.main INFO : Reading netlist: /home/spice/Desktop/ALIGN-public/examples/output_files_r_0_c_0/test_align/five_transistor_ota_Bulk.sp subckt=FIVE_TRANSISTOR_OTA_BULK, flat=0
2024-06-25 15:21:34 align.compiler.compiler INFO : Starting topology identification...
2024-06-25 15:21:35 align.compiler.user_const INFO : Reading constraint file: [PosixPath('/home/spice/Desktop/ALIGN-public/examples/output_files_r_0_c_0/test_align/five_transistor_ota_Bulk.const.json')]
2024-06-25 15:21:35 align.compiler.compiler INFO : Completed topology identification.
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 12
2024-06-25 15:21:35 align.cell_fabric.drc ERROR : Found errors: DRC 4
2024-06-25 15:21:36 align.pnr.main INFO : Running Place & Route for FIVE_TRANSISTOR_OTA_BULK
2024-06-25 15:21:36 align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/cmdline.py", line 197, in parse_args
    return schematic2layout(**vars(arguments))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/main.py", line 193, in schematic2layout
    variants = generate_pnr(topology_dir, primitive_dir, pdk_dir, pnr_dir, subckt, primitives=primitives, nvariants=nvariants, effort=effort,
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/pnr/main.py", line 285, in generate_pnr
    cap_map, cap_lef_s = cap_placer_driver(toplevel_args_d=toplevel_args_d, results_dir=None)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/pnr/cap_placer.py", line 13, in cap_placer_driver
    DB, verilog_d, fpath, opath, numLayout, effort = gen_DB_verilog_d(toplevel_args_d, results_dir)
                                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/pnr/build_pnr_model.py", line 204, in gen_DB_verilog_d
    DB, verilog_d = PnRdatabase( fpath, topcell, vfile, lfile, mfile, dfile, verilog_d_in=verilog_d_in, map_d_in=map_d_in, lef_s_in=lef_s_in)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/pnr/build_pnr_model.py", line 164, in PnRdatabase
    DB.ReadPDKJSON( path + '/' + drname)
RuntimeError: [json.exception.type_error.302] type must be number, but is null
```

I believe there may be errors in my [layers.json](https://github.com/Nakiana/demo/blob/master/layers.json) file, but I'm having trouble identifying them.

srini229, 2024-06-26:
Can you add the following to the `layers.json` under the `M1` layer `"GdsDataType"`:
```                
"Label": 4,
"Blockage": 5
```

Nakiana, 2024-06-27:
@srini229 ALIGN generates the GDS file, but there are still some DRC errors in the process. I modified the 'm' parameter in the `five_transistor_ota_Bulk.sp` file to `m=1` to speed up the results from ALIGN."
```
.subckt five_transistor_ota_Bulk vbias vss vdd von vin vip
mn1 tail vbias vss vss n w=400e-9 l=60e-9 nf=2 m=1
mn2 von vin tail vss n w=400e-9 l=60e-9 nf=2 m=1
mn3 vop vip tail vss n w=400e-9 l=60e-9 nf=2 m=1
mp4 von vop vdd vdd p w=400e-9 l=60e-9 nf=2 m=1
mp5 vop vop vdd vdd p w=400e-9 l=60e-9 nf=2 m=1
.ends five_transistor_ota
```
Below is the log:
```
2024-06-27 15:09:42 align.main INFO : Reading netlist: /home/spice/Desktop/ALIGN-public/examples/five_transistor_ota_Bulk/five_transistor_ota_Bulk.sp subckt=FIVE_TRANSISTOR_OTA_BULK, flat=0
2024-06-27 15:09:42 align.compiler.compiler INFO : Starting topology identification...
2024-06-27 15:09:43 align.compiler.user_const INFO : Reading constraint file: [PosixPath('/home/spice/Desktop/ALIGN-public/examples/five_transistor_ota_Bulk/five_transistor_ota_Bulk.const.json')]
2024-06-27 15:09:43 align.compiler.compiler INFO : Completed topology identification.
2024-06-27 15:09:43 align.cell_fabric.drc ERROR : Found errors: DRC 1
2024-06-27 15:09:43 align.cell_fabric.drc ERROR : Found errors: DRC 8
2024-06-27 15:09:43 align.cell_fabric.drc ERROR : Found errors: DRC 2
2024-06-27 15:09:43 align.pnr.main INFO : Running Place & Route for FIVE_TRANSISTOR_OTA_BULK
2024-06-27 15:09:43 align.pnr.build_pnr_model INFO : Reading contraint json file FIVE_TRANSISTOR_OTA_BULK.pnr.const.json
2024-06-27 15:09:43 align.pnr.build_pnr_model INFO : Reading contraint json file FIVE_TRANSISTOR_OTA_BULK.pnr.const.json
2024-06-27 15:09:43 align.pnr.placer INFO : Starting bottom-up placement on FIVE_TRANSISTOR_OTA_BULK 0
2024-06-27 15:09:43 PnR.placer.SeqPair.SeqPair INFO : Enumerated search
2024-06-27 15:09:43 PnR.placer.Placer.PlacementCoreAspectRatio_ILP INFO : Exhausted all permutations of seq pairs and found 1 placement solution(s)
2024-06-27 15:09:44 align.pnr.build_pnr_model INFO : Reading contraint json file FIVE_TRANSISTOR_OTA_BULK.pnr.const.json
2024-06-27 15:09:44 PnR.placer.SeqPair.SeqPair INFO : Enumerated search
2024-06-27 15:09:44 align.pnr.router INFO : Starting top_down routing on FIVE_TRANSISTOR_OTA_BULK 0 restricted to None
2024-06-27 15:09:44 PnR.router.Router.RouteWork INFO : GcellGlobalRouter: FIVE_TRANSISTOR_OTA_BULK
2024-06-27 15:09:44 PnR.router.GlobalGraph.dijkstra WARNING : Router-Warning: feasible path might not be found
2024-06-27 15:09:44 PnR.router.GlobalGraph.dijkstra WARNING : Router-Warning: feasible path might not be found
2024-06-27 15:09:44 PnR.router.GlobalGraph.dijkstra WARNING : Router-Warning: feasible path might not be found
2024-06-27 15:09:44 PnR.router.GlobalGraph.dijkstra WARNING : Router-Warning: feasible path might not be found
2024-06-27 15:09:44 PnR.router.GlobalGraph.dijkstra WARNING : Router-Warning: feasible path might not be found
2024-06-27 15:09:44 PnR.router.Router.RouteWork INFO : GcellDetailRouter: FIVE_TRANSISTOR_OTA_BULK
2024-06-27 15:09:44 PnR.router.Router.RouteWork INFO : Create power grid: FIVE_TRANSISTOR_OTA_BULK
2024-06-27 15:09:44 PnR.router.Router.RouteWork INFO : Power routing FIVE_TRANSISTOR_OTA_BULK
2024-06-27 15:09:44 align.cell_fabric.remove_duplicates ERROR : Found errors: SHORT: 0 OPEN: 4 DIFFERENT WIDTH: 68
2024-06-27 15:09:44 align.cell_fabric.drc ERROR : Found errors: DRC 109
2024-06-27 15:09:44 align.pnr.main INFO : OUTPUT json at /home/spice/Desktop/ALIGN-public/align_test/su_mux_test/3_pnr/FIVE_TRANSISTOR_OTA_BULK_0.json
2024-06-27 15:09:44 align.pnr.main ERROR : 113 LVS / DRC errors found !!!
2024-06-27 15:09:44 align.pnr.main INFO : OUTPUT error file at /home/spice/Desktop/ALIGN-public/align_test/su_mux_test/3_pnr/FIVE_TRANSISTOR_OTA_BULK_0.errors
2024-06-27 15:09:44 align.pnr.main INFO : OUTPUT gds.json /home/spice/Desktop/ALIGN-public/align_test/su_mux_test/3_pnr/FIVE_TRANSISTOR_OTA_BULK_0.python.gds.json

```

Below is the` FIVE_TRANSISTOR_OTA_BULK_0.errors`:
```
OPEN ('VDD', [[('M1', [3955, 1120, 4045, 1680]), ('M1', [4205, 3720, 4295, 5480]), ('M1', [3705, 3720, 3795, 5480]), ('M1', [3455, 1120, 3545, 1680]), ('M1', [3205, 3720, 3295, 5480]), ('M2', [2575, 1350, 4925, 1450]), ('M2', [2575, 4950, 4925, 5050]), ('V1', [3950, 1350, 4050, 1450]), ('V1', [3450, 1350, 3550, 1450]), ('V1', [4200, 4950, 4300, 5050]), ('V1', [3700, 4950, 3800, 5050]), ('V1', [3200, 4950, 3300, 5050]), ('M3', [3350, 1320, 3450, 5080]), ('V2', [3350, 1350, 3450, 1450]), ('V2', [3350, 4950, 3450, 5050]), ('M4', [3040, 1350, 3560, 1450]), ('V3', [3350, 1350, 3450, 1450]), ('M5', [3150, -50, 3250, 1450]), ('V4', [3150, 1350, 3250, 1450]), ('V0', [3970, 1370, 4030, 1430]), ('V0', [4220, 4350, 4280, 4410]), ('V0', [3720, 4350, 3780, 4410]), ('V0', [3720, 4350, 3780, 4410]), ('V0', [3470, 1370, 3530, 1430]), ('V0', [3220, 4350, 3280, 4410])], [('M3', [3320, 1320, 3480, 1480])], [('M3', [3350, 1350, 3450, 1450])], [('M4', [3120, 1320, 3280, 1480])], [('M4', [3320, 1320, 3480, 1480])], [('M5', [3120, -80, 3280, 80]), ('M6', [-50, -50, 3250, 50]), ('V5', [3150, -50, 3250, 50])], [('M5', [3120, 1320, 3280, 1480])], [('M5', [3150, 1350, 3250, 2850]), ('M6', [-50, 2750, 3250, 2850]), ('V5', [3150, 2750, 3250, 2850])], [('M5', [3120, 2720, 3280, 2880])], [('M5', [3150, 2750, 3250, 5650]), ('M6', [-50, 5550, 3250, 5650]), ('V5', [3150, 5550, 3250, 5650])], [('M5', [3120, 5520, 3280, 5680])], [('M5', [3150, 5550, 3250, 8450]),  [... cut ...]
DRC ERROR Enclosure violation on M6-V5 for VDD: [3150, 5550, 4850, 5650] does not sufficiently surround [3150, 5550, 3250, 5650], 30
DRC ERROR Enclosure violation on M5-V5 for VDD: [3150, 8350, 3250, 11250] does not sufficiently surround [3150, 8350, 3250, 8450], 30
DRC ERROR Enclosure violation on M6-V5 for VDD: [3150, 8350, 4850, 8450] does not sufficiently surround [3150, 8350, 3250, 8450], 30
DRC ERROR Enclosure violation on M6-V5 for VDD: [3150, 11150, 4850, 11250] does not sufficiently surround [3150, 11150, 3250, 11250], 30

```
my [layers.json](https://github.com/Nakiana/demo/blob/master/layers.json) file

srini229, 2024-06-27:
Did you check the DRCs with a golden checker like Calibre?

Nakiana, 2024-06-27:
@srini229 No, I didn't. So there is small flaw in the json file,  I should use Calibre to check the DRCs and modify either the JSON file or the SPICE file  .

closed 2024-09-26 by Nakiana