ALIGN-analoglayout/ALIGN-public #1334: My circuit consists of many subcircuits that are in some library files. Can Align read these library files?

Nakiana, 2024-06-26:
My circuit consists of many subcircuits that are in some library files. Can Align read these library files? If not, what can I do to make Align read the circuit and subcircuits at the same time? I don't want to revise my circuit to consist of only leaf circuits, as this would be very difficult.

srini229, 2024-06-26:
ALIGN supports standard SPICE subcircuit syntax. Please refer to the example:
[examples/switched_capacitor_filter](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/examples/switched_capacitor_filter)

Nakiana, 2024-06-26:
@srini229 Okay, this is a good example. However, I have a similar problem. I added subcircuits to my top circuit file, but after running ALIGN, I get an error:
```
2024-06-26 17:13:31 align.main INFO : Reading netlist: /home/spice/Desktop/ALIGN-public/examples/output_files_r_0_c_0/test_align_sb/sb_mux.sp subckt=SB_MUX, flat=0
2024-06-26 17:13:31 align.compiler.compiler INFO : Starting topology identification...
2024-06-26 17:13:31 align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/cmdline.py", line 197, in parse_args
    return schematic2layout(**vars(arguments))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/main.py", line 167, in schematic2layout
    primitive_lib = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/compiler/compiler.py", line 27, in generate_hierarchy
    ckt_data, primitive_library = compiler_input(
                                  ^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/compiler/compiler.py", line 76, in compiler_input
    ckt_parser.parse(lines)
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/schema/parser.py", line 88, in parse
    self._dispatch(cache)
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/schema/parser.py", line 102, in _dispatch
    args, kwargs = self._decompose(cache)
                   ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/schema/parser.py", line 118, in _decompose
    assert all(x.type in ('NAME', 'NUMBER', 'EXPR', 'EQUALS') for x in cache), cache
AssertionError: [Token(type='NAME', value='ptran'), Token(type='NAME', value='n_in'), Token(type='NAME', value='n_out'), Token(type='NAME', value='n_gate'), Token(type='NAME', value='n_gnd'), Token(type='NAME', value='m1'), Token(type='NAME', value='n_in'), Token(type='NAME', value='n_gate'), Token(type='NAME', value='n_out'), Token(type='NAME', value='n_gnd'), Token(type='NAME', value='lvtnch'), Token(type='NAME', value='L'), Token(type='EQUALS', value='='), Token(type='NUMBER', value='60e-9'), Token(type='NAME', value='W'), Token(type='EQUALS', value='='), Token(type='NUMBER', value='45e-9'), Token(type='NAME', value='nf'), Token(type='EQUALS', value='='), Token(type='NUMBER', value='2'), Token(type='NAME', value='m'), Token(type='EQUALS', value='='), Token(type='NUMBER', value='16'), Token(type='DECL', value='.ends'), Token(type='NAME', value='ptran')]

```

This is the subckt causing the error:
```
.subckt ptran n_in n_out n_gate n_gnd 
m1 n_in n_gate n_out n_gnd lvtnch L=60e-9 W=45e-9 nf=2 m=16 

.ends ptran
```

This is my models.sp:
```
.model lvtnch nmos L=1 W=1 nfin=1 nf=1 m=1  stack=1 parallel=1
.model lvtpch pmos L=1 W=1 nfin=1 nf=1 m=1  stack=1 parallel=1
.model hvtnch nmos L=1 W=1 nfin=1 nf=1 m=1  stack=1 parallel=1
.model hvtpch pmos L=1 W=1 nfin=1 nf=1 m=1  stack=1 parallel=1
```
What's causing the error?

srini229, 2024-06-26:
You cannot have any space at the end of each line and no empty lines within a subcircuit. You need to fix your transistor width which is smaller than the length.

Nakiana, 2024-06-26:
@srini229 It works. Thank you for your help and for reminding me to fix the transistor width. I wanted to address the issue at some point, but I feel I will have more trouble. Thank you again.

closed 2024-06-26 by srini229

Nakiana, 2024-06-27:
@srini229 It seems there are too many subcircuits, and ALIGN cannot find the top circuit. If I place the top circuit at the beginning, ALIGN will not be able to find the subcircuits.
```
2024-06-27 15:42:09 align.main INFO : Reading netlist: /home/spice/Desktop/ALIGN-public/examples/output_files_r_0_c_0/test_align_sb/sb_mux.sp subckt=SB_MUX, flat=0
2024-06-27 15:42:09 align.compiler.compiler INFO : Starting topology identification...
2024-06-27 15:42:09 align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/cmdline.py", line 197, in parse_args
    return schematic2layout(**vars(arguments))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/main.py", line 167, in schematic2layout
    primitive_lib = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/compiler/compiler.py", line 27, in generate_hierarchy
    ckt_data, primitive_library = compiler_input(
                                  ^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/compiler/compiler.py", line 84, in compiler_input
    ckt_data = create_data.read_inputs(design_name)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spice/Desktop/ALIGN-public/general/lib/python3.12/site-packages/align/compiler/create_database.py", line 33, in read_inputs
    assert top_subckt, f"{name.upper()} not found in library {[e.name for e in self.lib]}"
AssertionError: SB_MUX not found in library ['NMOS', 'PMOS', 'CAP', 'RES', 'IND', 'LVTNCH', 'LVTPCH', 'HVTNCH', 'HVTPCH', 'PTRAN', 'PTRANP', 'TGATE', 'TGATE_LP', 'REST']

```

There is my source file: [sb_mux.sp](https://github.com/Nakiana/demo/blob/master/sb_mux.sp). Top circuit is end .

kkunal1408, 2024-06-29:
It is not just the top subcircuit that has not been read. None of your subcircuit definitions have been read here.  You should see all the subcircuit names in your netlist here ['NMOS', 'PMOS', 'CAP', 'RES', 'IND', 'LVTNCH', 'LVTPCH', 'HVTNCH', 'HVTPCH', 'PTRAN', 'PTRANP', 'TGATE', 'TGATE_LP', 'REST'].
Looks like there is some issue with [models file ](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/master/pdks/Bulk45nm_Mock_PDK/models.sp)

Nakiana, 2024-07-01:
@kkunal1408 Thank you for your help; it worked out. However, I'm facing a new issue. Below is the error log:
```
2024-07-01 10:59:06 align.main INFO : Reading netlist: /home/spice/Desktop/ALIGN-public/examples/output_files_r_0_c_0/test_align_sb/sb_mux.sp subckt=SB_MUX, flat=0
2024-07-01 10:59:06 align.compiler.compiler INFO : Starting topology identification...
2024-07-01 10:59:07 align.compiler.preprocess INFO : removing parallel instances ['XMPUP1'] and updating XMPUP0 parameters
2024-07-01 10:59:07 align.compiler.preprocess INFO : stacking ['XMNDOWN1', 'XMNDOWN2'] 2
2024-07-01 10:59:07 align.compiler.preprocess INFO : removing parallel instances ['XMPUP1'] and updating XMPUP0 parameters
2024-07-01 10:59:07 align.compiler.preprocess INFO : stacking ['XMNDOWN1', 'XMNDOWN2'] 2
2024-07-01 10:59:07 align.compiler.preprocess INFO : Flattening dummy hierarchy PTRAN
2024-07-01 10:59:07 align.compiler.preprocess INFO : Removing hierarchy PTRAN from ['PTRAN', 'PTRANP', 'TGATE', 'TGATE_LP', 'REST', 'INV', 'INV_LP', 'NAND2', 'NAND2_DECODE', 'NOR2_DECODE', 'NAND2_LP', 'NAND3', 'NAND3_DECODE', 'NAND3_LP', 'RAM_TGATE', 'RAM_TGATE_LP', 'SB_MUX_DRIVER', 'SB_MUX_OFF', 'SB_MUX_PARTIAL', 'SB_MUX_ON_MUX_ONLY', 'SB_MUX_ON', 'CB_MUX_DRIVER', 'CB_MUX_OFF', 'CB_MUX_PARTIAL', 'CB_MUX_ON_MUX_ONLY', 'CB_MUX_ON', 'LUT', 'LUT_A_DRIVER', 'LUT_A_DRIVER_NOT', 'LUT_B_DRIVER', 'LUT_B_DRIVER_NOT', 'LUT_C_DRIVER', 'LUT_C_DRIVER_NOT', 'LUT_D_DRIVER', 'LUT_D_DRIVER_NOT', 'LUT_A_DRIVER_LOAD', 'LUT_B_DRIVER_LOAD', 'LUT_C_DRIVER_LOAD', 'LUT_D_DRIVER_LOAD', 'FF', 'LOCAL_BLE_OUTPUT', 'GENERAL_BLE_OUTPUT', 'BLE_OUTPUTS', 'LUT_OUTPUT_LOAD', 'LOCAL_MUX_SENSE', 'LOCAL_MUX_OFF', 'LOCAL_MUX_PARTIAL', 'LOCAL_MUX_ON_MUX_ONLY', 'LOCAL_MUX_ON', 'LOCAL_ROUTING_WIRE_LOAD', 'LOCAL_BLE_OUTPUT_LOAD', 'GENERAL_BLE_OUTPUT_LOAD', 'ROUTING_WIRE_LOAD_TILE_1', 'ROUTING_WIRE_LOAD_TILE_2', 'ROUTING_WIRE_LOAD_TILE_3', 'ROUTING_WIRE_LOAD_TILE_4', 'ROUTING_WIRE_LOAD', 'SB_MUX', 'PTRAN_1', 'INV_1', 'PTRAN_2', 'PTRAN_3', 'INV_2', 'INV_3']
2024-07-01 10:59:07 align.compiler.preprocess INFO : Flattening dummy hierarchy PTRAN_1
2024-07-01 10:59:07 align.compiler.preprocess INFO : Removing hierarchy PTRAN_1 from ['PTRANP', 'TGATE', 'TGATE_LP', 'REST', 'INV', 'INV_LP', 'NAND2', 'NAND2_DECODE', 'NOR2_DECODE', 'NAND2_LP', 'NAND3', 'NAND3_DECODE', 'NAND3_LP', 'RAM_TGATE', 'RAM_TGATE_LP', 'SB_MUX_DRIVER', 'SB_MUX_OFF', 'SB_MUX_PARTIAL', 'SB_MUX_ON_MUX_ONLY', 'SB_MUX_ON', 'CB_MUX_DRIVER', 'CB_MUX_OFF', 'CB_MUX_PARTIAL', 'CB_MUX_ON_MUX_ONLY', 'CB_MUX_ON', 'LUT', 'LUT_A_DRIVER', 'LUT_A_DRIVER_NOT', 'LUT_B_DRIVER', 'LUT_B_DRIVER_NOT', 'LUT_C_DRIVER', 'LUT_C_DRIVER_NOT', 'LUT_D_DRIVER', 'LUT_D_DRIVER_NOT', 'LUT_A_DRIVER_LOAD', 'LUT_B_DRIVER_LOAD', 'LUT_C_DRIVER_LOAD', 'LUT_D_DRIVER_LOAD', 'FF', 'LOCAL_BLE_OUTPUT', 'GENERAL_BLE_OUTPUT', 'BLE_OUTPUTS', 'LUT_OUTPUT_LOAD', 'LOCAL_MUX_SENSE', 'LOCAL_MUX_OFF', 'LOCAL_MUX_PARTIAL', 'LOCAL_MUX_ON_MUX_ONLY', 'LOCAL_MUX_ON', 'LOCAL_ROUTING_WIRE_LOAD', 'LOCAL_BLE_OUTPUT_LOAD', 'GENERAL_BLE_OUTPUT_LOAD', 'ROUTING_WIRE_LOAD_TILE_1', 'ROUTING_WIRE_LOAD_TILE_2', 'ROUTING_WIRE_LOAD_TILE_3', 'ROUTING_WIRE_LOAD_TILE_4', 'ROUTING_WIRE_LOAD', 'SB_MUX', 'PTRAN_1', 'INV_1', 'PTRAN_2', 'PTRAN_3', 'INV_2', 'INV_3']
2024-07-01 10:59:07 align.compiler.preprocess INFO : Flattening dummy hierarchy REST
2024-07-01 10:59:07 align.compiler.preprocess INFO : Removing hierarchy REST from ['PTRANP', 'TGATE', 'TGATE_LP', 'REST', 'INV', 'INV_LP', 'NAND2', 'NAND2_DECODE', 'NOR2_DECODE', 'NAND2_LP', 'NAND3', 'NAND3_DECODE', 'NAND3_LP', 'RAM_TGATE', 'RAM_TGATE_LP', 'SB_MUX_DRIVER', 'SB_MUX_OFF', 'SB_MUX_PARTIAL', 'SB_MUX_ON_MUX_ONLY', 'SB_MUX_ON', 'CB_MUX_DRIVER', 'CB_MUX_OFF', 'CB_MUX_PARTIAL', 'CB_MUX_ON_MUX_ONLY', 'CB_MUX_ON', 'LUT', 'LUT_A_DRIVER', 'LUT_A_DRIVER_NOT', 'LUT_B_DRIVER', 'LUT_B_DRIVER_NOT', 'LUT_C_DRIVER', 'LUT_C_DRIVER_NOT', 'LUT_D_DRIVER', 'LUT_D_DRIVER_NOT', 'LUT_A_DRIVER_LOAD', 'LUT_B_DRIVER_LOAD', 'LUT_C_DRIVER_LOAD', 'LUT_D_DRIVER_LOAD', 'FF', 'LOCAL_BLE_OUTPUT', 'GENERAL_BLE_OUTPUT', 'BLE_OUTPUTS', 'LUT_OUTPUT_LOAD', 'LOCAL_MUX_SENSE', 'LOCAL_MUX_OFF', 'LOCAL_MUX_PARTIAL', 'LOCAL_MUX_ON_MUX_ONLY', 'LOCAL_MUX_ON', 'LOCAL_ROUTING_WIRE_LOAD', 'LOCAL_BLE_OUTPUT_LOAD', 'GENERAL_BLE_OUTPUT_LOAD', 'ROUTING_WIRE_LOAD_TILE_1', 'ROUTING_WIRE_LOAD_TILE_2', 'ROUTING_WIRE_LOAD_TILE_3', 'ROUTING_WIRE_LOAD_TILE_4', 'ROUTING_WIRE_LOAD', 'SB_MUX', 'INV_1', 'PTRAN_2', 'PTRAN_3', 'INV_2', 'INV_3']
2024-07-01 10:59:07 align.compiler.preprocess INFO : Flattening dummy hierarchy PTRAN_2
2024-07-01 10:59:07 align.compiler.preprocess INFO : Removing hierarchy PTRAN_2 from ['PTRANP', 'TGATE', 'TGATE_LP', 'INV', 'INV_LP', 'NAND2', 'NAND2_DECODE', 'NOR2_DECODE', 'NAND2_LP', 'NAND3', 'NAND3_DECODE', 'NAND3_LP', 'RAM_TGATE', 'RAM_TGATE_LP', 'SB_MUX_DRIVER', 'SB_MUX_OFF', 'SB_MUX_PARTIAL', 'SB_MUX_ON_MUX_ONLY', 'SB_MUX_ON', 'CB_MUX_DRIVER', 'CB_MUX_OFF', 'CB_MUX_PARTIAL', 'CB_MUX_ON_MUX_ONLY', 'CB_MUX_ON', 'LUT', 'LUT_A_DRIVER', 'LUT_A_DRIVER_NOT', 'LUT_B_DRIVER', 'LUT_B_DRIVER_NOT', 'LUT_C_DRIVER', 'LUT_C_DRIVER_NOT', 'LUT_D_DRIVER', 'LUT_D_DRIVER_NOT', 'LUT_A_DRIVER_LOAD', 'LUT_B_DRIVER_LOAD', 'LUT_C_DRIVER_LOAD', 'LUT_D_DRIVER_LOAD', 'FF', 'LOCAL_BLE_OUTPUT', 'GENERAL_BLE_OUTPUT', 'BLE_OUTPUTS', 'LUT_OUTPUT_LOAD', 'LOCAL_MUX_SENSE', 'LOCAL_MUX_OFF', 'LOCAL_MUX_PARTIAL', 'LOCAL_MUX_ON_MUX_ONLY', 'LOCAL_MUX_ON', 'LOCAL_ROUTING_WIRE_LOAD', 'LOCAL_BLE_OUTPUT_LOAD', 'GENERAL_BLE_OUTPUT_LOAD', 'ROUTING_WIRE_LOAD_TILE_1', 'ROUTING_WIRE_LOAD_TILE_2', 'ROUTING_WIRE_LOAD_TILE_3', [... cut ...]
  File "/home/spice/Desktop/ALIGN-public/pdks/Huahong_PDK/mos.py", line 29, in __init__
    assert exact_width % 2 ==0, f"Transistor width {exact_width} must be even"
AssertionError: Transistor width 119 must be even

```
 I believe there is no odd width in the SPICE file.
Attached are my [SPICE file](https://github.com/Nakiana/demo/blob/master/sb_mux.sp), [models.sp](https://github.com/Nakiana/demo/blob/master/models.sp), and [layers.json](https://github.com/Nakiana/demo/blob/master/layers.json) file.

Nakiana, 2024-07-02:
any ideas @kkunal1408

kkunal1408, 2024-07-08:
Can you use  ConfigureCompiler constraint https://align-analoglayout.github.io/ALIGN-public/notes/const.html to turn off the auto-constraint generation?

Nakiana, 2024-07-09:
@kkunal1408  Yes, I just tried it, but it didn't work out.

srini229, 2024-07-09:
@kkunal1408 I fixed the exact_width error. Floating point truncation was the issue. You can use the attached [`mos.py`](https://github.com/ALIGN-analoglayout/ALIGN-public/blob/fix/odd_width/pdks/Bulk65nm_Mock_PDK/mos.py) to get the fixes. Upon fixing this, the PnR Verilog parser is complaining about a `SameTemplate` constraint in the Verilog JSON file.

Nakiana, 2024-07-18:
@kkunal1408 What's new with the `SameTemplate` error?