ALIGN-analoglayout/ALIGN-public #1354: Installation Issues

Rosnel14, 2025-06-05:
Hi, 

I have been trying to install the latest release of ALIGN and after creating my virtual environment and installing dependencies, I run the following command: 
`pip install -v .`

However, when building the project the terminal returns the following error after failing to build the wheel for align: 
`CMake Error at _skbuild/linux-x86_64-3.10/cmake-build/_deps/json-src/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.`

My initial thoughts were to change my version of cmake in the virtual environment but no luck in that part. Any help would be appreciated.

hoangtnguyen-3702, 2025-07-02:
> This worked for me to get align to successfully build.
>
> image.png (view on web)
>
> However, I proceed to check the that the schematic2layout.py is working
> as normal so I run both commands in my terminal separately:
>
> schematic2layout.py ../examples/telescopic_ota/ -p
> ../pdks/FinFET14nm_Mock_PDK/
>
> and
>
> schematic2layout.py -h
>
> Both of these commands give me the following error about circular imports.
>
> Traceback (most recent call last):
> File "../ALIGN-public-master/general/bin/schematic2layout.py", line 2, in
> import align
> File "../ALIGN-public-master/align/*init*.py", line 6, in
> from .main import schematic2layout
> File "../ALIGN-public-master/align/main.py", line 14, in
> from .pnr import generate_pnr
> File "../ALIGN-public-master/align/pnr/*init*.py", line 1, in
> from .checkers import *
> File "../ALIGN-public-master/align/pnr/checkers.py", line 7, in
> from .router import NType
> File "../ALIGN-public-master/align/pnr/router.py", line 12, in
> from .. import PnR
> ImportError: cannot import name 'PnR' from partially initialized module
> 'align' (most likely due to a circular import)
> (../ALIGN-public-master/align/*init*.py)
>
> Is this still an issue with my installation or build method? Unsure how to
> proceed.

Hi there,

I also tried the schematic2layout.py command with the telescopic OTA as you
did and I am having the same problem. I have posted my problem (#1355) to
the ALIGN-public directory already and am waiting for others' support. You
can also keep track of my post so that you might have the same solution.

And by the way, are you also working on layout automation with the ALIGN
framework? We might connect and support each other then!

Thank you,
Hoang.

srini229, 2025-07-03:
Can you add the line `cmake_args.append("-DCMAKE_POLICY_VERSION_MINIMUM=3.5")` to the `setup.py` file line 47 as shown below and try the command `pip install -v .`

[image]

srini229, 2025-07-03:
Your addition has a typo. It should be `cmake_args`, not `cmake_arg`.

srini229, 2025-07-08:
If you using docker, there is a prebuilt image of ALIGN available on docker hub:
[https://hub.docker.com/r/darpaalign/align-public](https://hub.docker.com/r/darpaalign/align-public)