ALIGN-analoglayout/ALIGN-public #1355: ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import)

hoangtnguyen-3702, 2025-07-02:
Hello everyone, 

Currently, I am working on the layout automation using the ALIGN framework. I have completed the installation of ALIGN but I could not run their suggested example of building the layout for the telescopic OTA (as in their document in https://align-analoglayout.github.io/ALIGN-public/notes/installation.html) using the following two commands

mkdir work && cd work
schematic2layout.py ../examples/telescopic_ota -p ../pdks/FinFET14nm_Mock_PDK/

I had the error as below

ImportError: cannot import name 'PnR' from partially initialized module 'align' (most likely due to a circular import) (/mnt/c/Users/hoangtn/ALIGN_layout_automation/ALIGN-public/align/__init__.py).

I am using Ubuntu 22.04.5 LTS and Python 3.12.10 to run ALIGN. 

Can anyone help me solve my problem?

Thank you very much!

hoangtnguyen-3702, 2025-07-03:
Hello @srini229,

I am using Python 3.10.12, do I need to upgrade it to Python 3.12 to install all the wheels you provided?

Thank you,
Hoang.

hoangtnguyen-3702, 2025-07-03:
Hello @srini229,

While waiting, I have installed Python 3.12 to my Ubuntu 22.04.5, I added the set of wheels you provided and use the command pip install <downloaded directory>/*.whl and it worked. 

I can now run the example of the telescopic OTA and generate the .gds file. 

[image]

I still have several warning as in the figure above. I am not sure if these warning affect the output result or not?

Thank you so much for your support with the wheels directory. I think now I am good with using Python 3.12 and your provided wheels if the mentioned warning can be ignored. 

Best regards,
Hoang.

srini229, 2025-07-04:
For now, you can ignore these warnings. I will check them out.

hoangtnguyen-3702, 2025-07-04:
Hi @srini229,

Thank you very much for your support! Let me know if I need to fix anything to solve the warnings.

hoangtnguyen-3702, 2025-07-05:
> For now, you can ignore these warnings. I will check them out.

Dear @srini229,

Can you please have a look at my problem in generating the layout for the LNA circuit in the CircuitDatabase directory (mentioned in issue #1357)?

Thank you very much!

danbeibei26-prog, 2026-04-21:
> Can you install all the wheels in this attached directory in your Python virtualenv and see if it helps? [ALIGN_wheel](https://drive.google.com/drive/folders/1awpjcVuUPQxqQWup8SVJyQP--D_l8hY-)
> 
> I have built the ALIGN wheel for Python 3.12 on Ubuntu 22.04.5.
> 
> Download the entire set of wheels to a directory and use the command `pip install <downloaded directory>/*.whl`.

> I can build a wheel for your configuration if you can specify your OS and Python version.
I'm using Ubuntu 24.04.3 and the Python version is 3.10.20. Could you please assist me?