ALIGN-analoglayout/ALIGN-public #1357: AssertionError when running ALIGN with the LNA in the CircuitDatabase directory

hoangtnguyen-3702, 2025-07-04:
Hi all,

I have installed ALIGN and tried the example of the telescopic OTA successfully. However, when I try with the LNA circuit in the CircuitDatabase directory with the command (from the work directory)

schematic2layout.py ../CircuitsDatabase/Sized_Netlists/Wireless-Radio_frequency/LNA/ -p ../pdks/FinFET14nm_Mock_PDK/ 

the error is as follows:

align.cmdline ERROR : Fatal Error. Cannot proceed
Traceback (most recent call last):
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/cmdline.py", line 197, in parse_args
    return schematic2layout(**vars(arguments))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/main.py", line 167, in schematic2layout
    primitive_lib = generate_hierarchy(netlist, subckt, topology_dir, flatten, pdk_dir)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/compiler/compiler.py", line 27, in generate_hierarchy
    ckt_data, primitive_library = compiler_input(
                                  ^^^^^^^^^^^^^^^
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/compiler/compiler.py", line 76, in compiler_input
    ckt_parser.parse(lines)
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/schema/parser.py", line 88, in parse
    self._dispatch(cache)
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/schema/parser.py", line 106, in _dispatch
    self._process_instance(token.value.upper(), args, kwargs)
  File "/mnt/c/Users/hoangtn/ALIGN_layout_auto/align_py3.12/lib/python3.12/site-packages/align/schema/parser.py", line 161, in _process_instance
    assert len(args) == len(model.pins), \
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: Model NMOS has 4 pins ['D', 'G', 'S', 'B']. 59 nets ['NET011', 'VAUX', 'VDD', 'RESISTOR', 'XR4', 'NET063', 'SF_BIAS', 'VDD', 'RESISTOR', 'XR10', 'NET0252', 'VMAIN', 'VDD', 'RESISTOR', 'XR3', 'NET033', 'SF_BIAS', 'VDD', 'RESISTOR', 'XC12', 'NET0252', 'VSS', 'VSS', 'MIMCAP', 'XC9', 'NET0252', 'VSS', 'VSS', 'MIMCAP', 'XC6', 'OUTN', 'NET063', 'VDD', 'MIMCAP', 'XC43', 'IN_INT', 'NET011', 'VDD', 'MIMCAP', 'XC5', 'OUTP', 'NET033', 'VDD', 'MIMCAP', 'XC11', 'NET059', 'VDD', 'VDD', 'MIMCAP', 'XC8', 'NET047', 'VDD', 'VDD', 'MIMCAP', 'XM10', 'VDD', 'NET033', 'VOUTP', 'VOUTP'] were passed when instantiating XR5.

Is that because the netlist file LNA_QM.sp should be changed? Can you please give me a solution to this?

The netlist of the LNA circuit in the LNA_QM.sp file is as below.

[image]

Thank you very much,
Hoang.

hoangtnguyen-3702, 2025-07-08:
> Inductor layouts are not supported natively in ALIGN. We use a blackbox flow to import kit inductors. This flow is experimental though.

Hello @srini229,

Can you provide more information about the blackbox flow to import kit inductors you mentioned? 

And sorry I am not sure if I understand correctly what you mean by experimental flow, is this blackbox flow still under development by ALIGN?

Thank you very much!
Hoang

srini229, 2025-07-08:
The experimental flow isn't properly integrated into ALIGN and isn't guaranteed to work properly.

You can look at a working example in the directory: `tests/gdsconv/black_box_test`.

hoangtnguyen-3702, 2025-07-08:
> Thank you for your provided directory [@srini229](https://github.com/srini229).
> 
> Regarding resistors, does ALIGN support 3-terminal resistors or just 2-terminal resistors?
> 
> Thank you, Hoang.

I mean both resistor and capacitor for my question, does ALIGN only support 2-terminal resistors and capacitors instead of 3-terminal?

srini229, 2025-07-10:
The default generator does only two terminal passives. You can modify them to generate three terminals.

hoangtnguyen-3702, 2025-07-10:
> The default generator does only two terminal passives. You can modify them to generate three terminals.

Do you have any instructions on this modification to 3-terminal devices?

Thank you!